ARMmbed/mbed-os #830: RTOS: Cortex A - stack larger than 64k

0xc0170, 2015-01-05:
The version for Cortex-M was fixed via #826. It should be also applied to Cortex A version.

Snippet from P_TCB:
```
  /* Hardware dependant part: specific for Cortex processor                  */
  U8     stack_frame;             /* Stack frame: 0x1 Basic/Extended, 0x2 FP stacked/not stacked */
  U8     reserved;
  U16    priv_stack;              /* Private stack size, 0= system assigned  */
  U32    tsk_stack;               /* Current task Stack pointer (R13)        */
  U32    *stack;                  /* Pointer to Task Stack memory block      */

  /* Task entry point used for uVision debugger                              */
  FUNCP  ptask;                   /* Task entry address                      */
} *P_TCB;
```

priv_stack is there U16.

0xc0170, 2015-01-05:
Fixed via #826

adamgreen, 2015-01-05:
@0xc0170 I thought you opened this bug to track the propagating of change 826 to the Cortex-A branch of the RTX code?

TomoYamanaka, 2015-07-22:
Hello,

Thanks for your comment.
We have a plan to modify this problem in Cortex-A.
We will update RTOS to support NEON.
We will modify both issues in the same pull request.

TomoYamanaka, 2015-10-01:
Hello,

We updated RTOS to support NEON.
(https://github.com/mbedmicro/mbed/pull/1326)
We have modified this issues in the update.

0xc0170, 2015-10-01:
@TomoYamanaka Means this should be fixed on master  ,and this can be closed?

TomoYamanaka, 2015-10-01:
Hello,

Yes, This thread can be closed since the fix of this isuue was reflected on master.
Please close this thread.

0xc0170, 2015-10-01:
OK, I checked, should be fixed. Thanks

closed 2015-10-01 by 0xc0170