ARMmbed/mbed-os #841: FRDM-K22F: When using I2C there is no console output on CDC

PrzemekWirkus, 2015-01-12:
* I've ran I2C tests with 24LC256 (EEPROM) and I notices no printouts are send via virtual serial port.
* I can see when I run tests, they pass (red LED blink with "OK" low frequency) but no output on USB serial port.
* I've checked with logical analyzer I2C, it looks OK (e.g. frequency was 400kHz and transmission was uninterrupted).
```
singletest.py -i test_spec.json -M muts_all.json -P -V
```
```
Building library CMSIS (K22F, ARM)
Building library MBED (K22F, ARM)
Building project I2C_EEPROM (K22F, ARM)
Compile: main.cpp
Link: i2c_eeprom
Elf2Bin: i2c_eeprom
Executing 'python host_test.py -d G:\ -f "C:\Work\mbed\build\test\K22F\ARM\MBED_A19\i2c_eeprom.bin" -p COM93 -t 15 -C 4 -m K22F'
Test::Output::Start
MBED: Instrumentation: "COM93" and disk: "G:\"
HOST: Copy image onto target...
HOST: Initialize serial port...
HOST: Reset target...
Test::Output::Finish
TargetTest::K22F::ARM::MBED_A19::I2C EEPROM read/write test [TIMEOUT] in 30.95 of 15 sec
Building project I2C_EEPROM_LINE (K22F, ARM)
Compile: main.cpp
Link: i2c_eeprom_line
Elf2Bin: i2c_eeprom_line
Executing 'python host_test.py -d G:\ -f "C:\Work\mbed\build\test\K22F\ARM\MBED_A25\i2c_eeprom_line.bin" -p COM93 -t 10 -C 4 -m K22F'
Test::Output::Start
MBED: Instrumentation: "COM93" and disk: "G:\"
HOST: Copy image onto target...
HOST: Initialize serial port...
HOST: Reset target...
Test::Output::Finish
TargetTest::K22F::ARM::MBED_A25::I2C EEPROM line read/write test [TIMEOUT] in 20.96 of 10 sec
Test summary:
+---------+--------+-----------+----------+---------------------------------+--------------------+---------------+-------+
| Result  | Target | Toolchain | Test ID  | Test Description                | Elapsed Time (sec) | Timeout (sec) | Loops |
+---------+--------+-----------+----------+---------------------------------+--------------------+---------------+-------+
| TIMEOUT | K22F   | ARM       | MBED_A19 | I2C EEPROM read/write test      |       30.95        |       15      |  0/1  |
| TIMEOUT | K22F   | ARM       | MBED_A25 | I2C EEPROM line read/write test |       20.96        |       10      |  0/1  |
+---------+--------+-----------+----------+---------------------------------+--------------------+---------------+-------+
Result: 2 TIMEOUT

Completed in 55.51 sec
```

Sissors, 2015-01-13:
@0xc0170 promised me a hug if I would solve this, and I am desperate for hugs :(.

Pinout:
// USB Pins 
USBTX = PTE0, 
USBRX = PTE1, 

Test pinout:
#elif defined(TARGET_K22F) 
I2C i2c(PTE0, PTE1); 

Woohoo, free hug. (In other words: Use different pins, A4, A5 which are the Arduino I2C pins should work probably).

0xc0170, 2015-01-13:
Thanks Erik for looking at it. We should update I2C tests to include other pins, as obviously they overlap with USBxX

closed 2015-01-26 by 0xc0170

PrzemekWirkus, 2015-01-28:
No free hugs

```c
#elif defined(TARGET_K64F)
I2C i2c(PTE25, PTE24);

#elif defined(TARGET_K22F)
I2C i2c(PTE0, PTE1);

#elif defined(TARGET_K20D50M)
I2C i2c(PTB3, PTB2);
```

Pin out is set in test for K22F.

0xc0170, 2015-01-28:
@PrzemekWirkus those 2 are used by serial object as TX and RX therefore it fails to print anything