ARMmbed/mbed-os #843: linking esb_gcc.a (nrf51822 enhanced shock burst) with mbed

freber, 2015-01-14:
Gentlemen,

Love your work and trying to get nrf51822 esb (enhanced shock burst) to work with mbed. When linking gcc_esb.a (the esb library) there is a conflict that I would like to get help with.

```
Fredrik:Build freber$ make
Scanning dependencies of target MY_APP.elf
[  1%] Building C object CMakeFiles/MY_APP.elf.dir/mbed-src/libraries/mbed/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/app_common/app_timer.c.o
In file included from /Users/freber/dev/myApp/mbed-src/libraries/mbed/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/app_common/app_timer.c:19:0:
/Users/freber/dev/myApp/nRF51822/nordic/nrf-sdk/nrf_delay.h:48:1: warning: 'inline' is not at beginning of declaration [-Wold-style-declaration]
 static void __INLINE nrf_delay_us(uint32_t volatile number_of_us)
 ^
Linking CXX executable MY_APP.elf
../nRF51822/nordic/nrf-sdk/Lib/esb/gcc/esb_gcc.a(nrf_notification_manager.o): In function `SWI0_IRQHandler':
nrf_notification_manager.c:(.text+0xac): multiple definition of `SWI0_IRQHandler'
CMakeFiles/MY_APP.elf.dir/mbed-src/libraries/mbed/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/app_common/app_timer.c.o:app_timer.c:(.text.SWI0_IRQHandler+0x0): first defined here
../nRF51822/nordic/nrf-sdk/Lib/esb/gcc/esb_gcc.a(nrf_flywheel.o): In function `TIMER2_IRQHandler':
nrf_flywheel.c:(.text+0x1ec): multiple definition of `TIMER2_IRQHandler'
CMakeFiles/MY_APP.elf.dir/mbed-src/libraries/mbed/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/pwmout_api.c.o:pwmout_api.c:(.text.TIMER2_IRQHandler+0x0): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [MY_APP.elf] Error 1
make[1]: *** [CMakeFiles/MY_APP.elf.dir/all] Error 2
make: *** [all] Error 2
```

0xc0170, 2015-01-14:
Hi,

based on the linking error, pwm in mbed is implemented using the Timer2 IRQ. Although mbed uses NVIC reallocation (=dynamic vectors) , thus this should be changed as other platforms have it.

Can you send pull request fixing this Timer IRQ in pwmout code file?

0xc0170, 2015-01-26:
@freber  Any progress?

freber, 2015-02-05:
I'm sorry but I'm really bad with git. I managed to change SWI0_IRQHandler to SWI2_IRQHandler and got it to link without conflict. What is the process of doing the pull request?

Sorry for late answer.

jrobeson, 2015-10-18:
if it's just one file, then click edit on the file on this web interface. It will fork the repository (to your username) and then once you save it, it will allow you to submit a pull request.

jrobeson, 2017-07-18:
@freber : did you end up with a working application from this?

freber, 2017-07-18:
Yes Sir!

This was some time ago but it works now.

Best regards,

closed 2017-07-18 by 0xc0170