ARMmbed/mbed-os #856: RTOS Compiler Warning

neilt6, 2015-01-21:
Compiling the RTOS library in the online compiler is currently issuing the following compiler warning:

```
Warning: A value of type "unsigned char *" cannot be assigned to an entity of type "uint32_t *" in "mbed-rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h", Line: 319, Col: 38
```

I believe this may be a result of pull request #812, which changed `osThreadDef_t.stack_pointer` to a `uint32_t*` type.

adamgreen, 2015-01-21:
Maybe this will fix it?
``` diff
diff --git a/libraries/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h b/libraries/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h
index b0e8806..e7e5311 100755
--- a/libraries/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h
+++ b/libraries/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h
@@ -304,10 +304,10 @@ osThreadDef_t os_thread_def_main = {(os_pthread)main, osPriorityNormal, 0, NULL}
 #endif
 
 #ifdef __CC_ARM
-extern unsigned char     Image$$RW_IRAM1$$ZI$$Limit[];
+extern uint32_t          Image$$RW_IRAM1$$ZI$$Limit[];
 #define HEAP_START      (Image$$RW_IRAM1$$ZI$$Limit)
 #elif defined(__GNUC__)
-extern unsigned char     __end__[];
+extern uint32_t          __end__[];
 #define HEAP_START      (__end__)
 #elif defined(__ICCARM__)
 #pragma section="HEAP"
```

neilt6, 2015-01-21:
Seems to work ok on my end

adamgreen, 2015-01-22:
Thanks for testing.  I have submitted pull request #857 with that change.

closed 2015-01-22 by 0xc0170