ARMmbed/mbed-os #859: Change Thread ctor stack pointer to uint32_t*

neilt6, 2015-01-23:
As a followup to pull request #812, I believe the stack pointer in the Thread constructor should be changed to accept a `uint32_t*` instead of an `unsigned char*`. Reason being is I manually define my own stacks, and I have to make sure they're word-aligned to avoid nasty surprises like so:
``` C++
unsigned char threadStack[1024] __attribute__((aligned));
```
Not only is `__attribute__((aligned))` not portable, but I would venture a guess that the average mbed developer has no idea what it does or why it's necessary is this situation. Changing the constructor will force these developers to properly align their manually defined stacks without requiring an explanation.

PrzemekWirkus, 2015-01-30:
@neilt6: I think if you want to be portable you can do something "dirty" like this:
```C++
unsigned char threadStack[1024];  // Replace 1024 with 1028 to get full 1kB of stack ;)
unsigned char* threadStackPtr = (threadStack | 0x03);  // Align to 4 up
int threadStackSize = 1024 - (threadStackPtr - threadStack);

// Create thread somewhere
Thread thread = Thread(threadStackPtr, threadStackSize);
```

neilt6, 2015-01-30:
@PrzemekWirkus, while this would certainly work, the average mbed developer will not know to do this either. Changing the constructor forces developers to align their stacks.

0xc0170, 2016-10-28:
@c1728p9 Sounds reasonable to update the stack pointer argument?

c1728p9, 2016-10-28:
This might be a breaking API change, since you get a compiler error if the type is wrong. What about modifying the alignment inside the thread class rather than making a user do it? RTX already does this to some extent to force 8 byte alignment.

0xc0170, 2016-10-31:
> What about modifying the alignment inside the thread class rather than making a user do it?

Sounds reasonable. @neilt6 What do you think?

neilt6, 2016-10-31:
Sounds good to me.

0xc0170, 2017-10-12:
Is this still relevant @neilt6 . Thread implementation contains this line https://github.com/ARMmbed/mbed-os/blob/master/rtos/Thread.cpp#L47 , that is relevant for this issue.

neilt6, 2017-10-13:
@0xc0170 I don't believe casting the stack pointer to a `uint32_t*` will correct the alignment, so the issue is still relevant.

pan-, 2017-10-15:
Agreed, it has not been fix and a cast won't solve the issue. However we need the max alignment available on the target, `uint32_t` does not offer that guarantee.

SenRamakri, 2018-07-18:
As of now, Thread.cpp constructor implementation ensures that stack is 8-byte aligned. So, closing this issue as the original need to align to stack by using uint32 is no longer needed.

closed 2018-07-18 by SenRamakri