ARMmbed/mbed-os #871: I2C start for LPC812 (and LPC824)

Willem23, 2015-01-27:
@K4zuki @0xc0170

I agree with @K4zuki that the same patch is needed for LPC824 sources. I had
a quick look at that platform and its I2C architecture looks the same.
However, latest mbed source is different cause it uses the ROM code for the
LPC824 i2c lib. I have not yet tested that on the LPC812. I also don't have
the LPC824 so cant test it anyhow.

The proposal for the new i2c.start() is listed below. I
think we could also improve the repeated_start flag: Set it when a Start is
generated, reset it when a Stop is generated. That way the state will be
correct when Stop is called due to some error + abort in any of the other
i2c functions. Question is do you want to support the new start for other
platforms too. What about breaking the API by removing the old i2c.start()
which is wrong for at least some platforms.

/** Creates a start condition on the I2C bus  */
inline int i2c_start_addr(i2c_t *obj, int address_rw, int repeat)  {
    int status;

    //Store the address_rw and then generate STA
    I2C_DAT(obj) = address_rw;  

//Repeated Start when previous Start was not followed by a Stop or when
//forced to perform   Repeated Start
    if (repeated_start || repeat) {
        // STA bit and Continue bit to complete previous RD or WR
        obj->i2c->MSTCTL = (1 << 1) | (1 << 0);
    }
    else {
        obj->i2c->MSTCTL = (1 << 1); // STA bit
    }

    // wait for it to complete
    i2c_wait_SI(obj);

    status = i2c_status(obj);    
    if ((status == 0x01) || (status == 0x02)) { // RX RDY or TX RDY
      // Start succeeded, Slave ACK
      repeated_start = 1; // remember that bus is now in use
      return 1;
    }
    else {
      // Start failed, Slave NACK, release bus, or do we want caller to issue Stop?
      // i2c_stop(obj);
      return 0;
    }  
}

//Generate Stop condition and wait until bus is Idle
//Will also send NAK for previous RD
inline int i2c_stop(i2c_t *obj) {
    int timeout = 0;

// STP bit and Continue bit. Sends NAK to complete previous RD
    obj->i2c->MSTCTL = (1 << 2) | (1 << 0); 
    
   //Spin until Ready (b0 == 1) and Status is Idle (b3..b1 == 000)
    while ((obj->i2c->STAT & ((7 << 1) | (1 << 0))) != ((0 << 1) | (1 << 0)))  { 
        timeout ++;
        if (timeout > 100000) return 1;
    }

    repeated_start = 0; // bus free
    return 0;
}

Sissors, 2015-01-27:
I have a different proposal on this topic, which is very likely not going to happen, but I am going to propose it anyway: Remove the I2C start, stop and single byte read/write functions, and just force everyone to use the complete functions. Most hardware was intended for this (example is the Freescale platform where we have to do weird stuff to make byte read work), and I don't see a reason  to maintain this duplicate set of commands.

Are there really a significant number of situations where those functions are needed, and aren't just confusing people?

sg-, 2015-01-27:
@Sissors +1

mbednotifications, 2015-01-28:
Hi Erik, Sam,

 

I know some people say we should get rid of the I2C byte operations
altogether. That may be a solution for some, but I think the single start,
write, read and stop are of use when you need to experiment or work with
slaves that don't play nice. There are for example LCD controllers (chip on
glass) that can't or don't send any ACKs. The blockwrites currently abort
communications when there is no ACK. So when you get rid of the single
operations we need at least a version of the block operations that ignore
all NACKed messages and press on.

That and the fact that we break the API makes me prefer to keep the single
ops with a WARNING that it may not always work as you would like and add the
new start function with the address as a parameter. 

 

Regards,

Wim

 

  _____  

From: mbed-devel@googlegroups.com [mailto:mbed-devel@googlegroups.com] On
Behalf Of Sissors
Sent: dinsdag 27 januari 2015 23:17
To: mbedmicro/mbed
Subject: Re: [mbed] I2C start for LPC812 (and LPC824) (#871)

 

I have a different proposal on this topic, which is very likely not going to
happen, but I am going to propose it anyway: Remove the I2C start, stop and
single byte read/write functions, and just force everyone to use the
complete functions. Most hardware was intended for this (example is the
Freescale platform where we have to do weird stuff to make byte read work),
and I don't see a reason to maintain this duplicate set of commands.

Are there really a significant number of situations where those functions
are needed, and aren't just confusing people?

-
<https://github.com/mbedmicro/mbed/issues/871#issuecomment-71739756>  it on
GitHub.
<https://github.com/notifications/beacon/AFAC15n8_NSoWr-CkvXhFW4zWWn0vJcCks5
nmAXNgaJpZM4DX-t9.gif> 

You received this message because you are subscribed to the Google Groups
"mbed-devel" group.
To unsubscribe from this group and stop receiving emails from it, send an
email to mbed-devel+unsubscribe@googlegroups.com.
For more options, visit https://groups.google.com/d/optout.

Sissors, 2015-01-28:
While I see your point, I do wonder how much the I2C API should provide for non-standard I2C. For those few also for example a software bitbanging setup could maybe be used.

Specifically regarding adding a new start function, are there any other non-standard situations where that would be a problem? So where you send a start, and then either you don't want to send a byte, or you don't know yet which byte to send, or possibly there are weird timing demands which require a delay before sending the first byte. That would be possible with the old API, but not with this one.

mbednotifications, 2015-01-29:
I am sure there are plenty special cases where there might be desire for
more control when sending Start. Problem is that in case of the LPC800 you
don't have any choice. Generating a Start will AUTOMATICALLY kick the I2C
engine to send out whatever data is in the DAT register which is assumed to
be the address whether you like it or not.

It even continues to AUTOMATICALLY read the first byte from the Slave when
the address is for a read operation and was ACKed. 

So I am not proposing to delete the original i2c.start, just to add an
additional function that at least allows you to send out a correct Start +
Address. The original function might work for the lpc812 but it will work
for the lpc1768 and others. The new function will work for all.

Repeated calls to start, stop or even to toggle the clock are sometimes used
and recommended to recover from bus errors. So I think they should remain,
but must always be used with caution. Preferred approach should be to use
the block operations.

 

  _____  

From: mbed-devel@googlegroups.com [mailto:mbed-devel@googlegroups.com] On
Behalf Of Sissors
Sent: woensdag 28 januari 2015 22:20
To: mbedmicro/mbed
Cc: mbednotifications
Subject: Re: [mbed] I2C start for LPC812 (and LPC824) (#871)

 

While I see your point, I do wonder how much the I2C API should provide for
non-standard I2C. For those few also for example a software bitbanging setup
could maybe be used.

Specifically regarding adding a new start function, are there any other
non-standard situations where that would be a problem? So where you send a
start, and then either you don't want to send a byte, or you don't know yet
which byte to send, or possibly there are weird timing demands which require
a delay before sending the first byte. That would be possible with the old
API, but not with this one.

-
<https://github.com/mbedmicro/mbed/issues/871#issuecomment-71918238>  it on
GitHub.
<https://github.com/notifications/beacon/AFAC19oXOD3IN3X2LicsjncTPzypZV_Gks5
nmUnrgaJpZM4DX-t9.gif> 

You received this message because you are subscribed to the Google Groups
"mbed-devel" group.
To unsubscribe from this group and stop receiving emails from it, send an
email to mbed-devel+unsubscribe@googlegroups.com.
For more options, visit https://groups.google.com/d/optout.

closed 2017-10-27 by ChiefBureaucraticOfficer