ARMmbed/mbed-os #886: [NRF51822][BLE Nano] I2C Issue

freber, 2015-02-05:
I'm struggling to get I2C to work with the nRF51822 (BLE Nano).
I've tried to setup the I2C pins also on the standard pinout (http://static1.squarespace.com/static/5039e08be4b00cf0e8cf88cd/t/543f6b58e4b0847bb28660f8/1413442404211/?format=1500w) but without any success. It feels like I'm missing something simple here but can't figure it out.

I'm on @adamgreen's  repo (gcc4mbed) and is currently running rev 92.

Below is my isolated test case:
```
#include "mbed.h"

#define TMP007_I2CADDR 0x40
#define TMP007_DEVID 0x1F

I2C i2c(P0_29, P0_28); // SDA, SCL
// I2C i2c(P0_10, P0_8); // SDA, SCL

Serial pc(USBTX, USBRX); // tx, rx

int main(void) {

  char temp_write[3] = {0};
  char temp_read[2] = {0};

	pc.printf("start!\n");

  // Set up what register to read from
  temp_write[0] = TMP007_DEVID;
  i2c.start();
  i2c.write(TMP007_I2CADDR, temp_write, 1, false);
  i2c.stop();

  // Read actual data
  i2c.start();
  i2c.read(TMP007_I2CADDR, temp_read, 2, 0);
  i2c.stop();

  pc.printf("Returned: 0x%X%X\n", temp_read[0], temp_read[1]);
  pc.printf("Should return: 0x%X\n", TMP007_DEVID);

}

```

Output:

```
start!
Returned: 0x00
Should return: 0x1F
```

Sissors, 2015-02-07:
The i2c complete read/write functions, which you are using, already include start and stop commands, you should not add them again.

0xc0170, 2015-02-07:
@freber please read the documentation, for I2C master: https://github.com/mbedmicro/mbed/blob/master/libraries/mbed/api/I2C.h

Can you report the result if you remove them as Erik suggested?

freber, 2015-02-09:
Hi,

I've tried both with and without the start and stop call. I will continue to investigate the issue. Tried with BMD-200 (Rigado) and it works fine so the nRF51822 as such is working. Must be some pin mapping. How is the specific board targeted? I can see that there is a "sub" target namned TARGET_RBLAB_BLENANO.

best regards,
Fredrik

Willem23, 2015-02-09:
Quick check of the datasheet for the TMP007 shows that the slaveaddress is 0x80 when both ADR1 and ADR0 are low. The mbed I2C libs use the 8bit format rather than the 7bit format that your program uses. So address should be 0x80 instead of 0x40 (see table 2). See also fig 23 for the timing (and notice that the slave address is wrong in frame 1 and frame 2!).

Also the default value in the DevID register is 0x0078 (see table 5)

#include "mbed.h"

#define TMP007_I2CADDR 0x80
#define TMP007_DEVID      0x1F

I2C i2c(P0_29, P0_28); // SDA, SCL
// I2C i2c(P0_10, P0_8); // SDA, SCL

Serial pc(USBTX, USBRX); // tx, rx

int main(void) {
  char temp_write[3] = {0};
  char temp_read[2] = {0};

    pc.printf("start!\n");

  // Set up what register to read from
  temp_write[0] = TMP007_DEVID;
  i2c.write(TMP007_I2CADDR, temp_write, 1);

  // Read actual data
  i2c.read(TMP007_I2CADDR, temp_read, 2);

  pc.printf("Returned: 0x%X%X\n", temp_read[0], temp_read[1]);
  pc.printf("Should return: 0x0078\n");
}

FixedFilip, 2016-05-10:
Hi everyone!

I've got a new RedBear BLE Nano and I'm trying to get the current temperature with a TMP007 as well but using the code above (with I2C i2c(P0_10, P0_8); // SDA, SCL instead of I2C i2c(P0_29, P0_28);) I get as response 0x00 instead of 0x0078. 
BLE's P0_10 is connected to TMP's SDA with 10k pullup and P0_8 to SCL with 10k pullup, too. 

Does anybody know how to fix it or can anybody tell me what's maybe the mistake? 

`#include "mbed.h"

#define TMP007_I2CADDR 0x80
#define TMP007_DEVID 0x1F

//I2C i2c(P0_29, P0_28); // SDA, SCL
 I2C i2c(P0_10, P0_8); // SDA, SCL

Serial pc(USBTX, USBRX); // tx, rx

int main(void)
{
    char temp_write[3] = {0};
    char temp_read[2] = {0};
    pc.baud(9600);
    wait_ms(5);
    pc.printf("start!\n");

    while(1) {
// Set up what register to read from
        temp_write[0] = TMP007_DEVID;
        i2c.write(TMP007_I2CADDR, temp_write, 1);

// Read actual data
        i2c.read(TMP007_I2CADDR, temp_read, 2);

        pc.printf("Returned: 0x%X%X\n", temp_read[0], temp_read[1]);
        pc.printf("Should return: 0x0078\n");
        wait_ms(1000);
    }
}`

Thank you very much!

Filip

closed 2017-10-27 by ChiefBureaucraticOfficer