ARMmbed/mbed-os #888: I2C busreset

frankvnk, 2015-02-06:
### Adding I2C busreset
I attempted to make my [KL25Z busreset code](http://developer.mbed.org/users/frankvnk/code/KL25Z_I2C_busreset/) generic but quickly found out it was going to be large and slow : a generic function for finding out exactly what pins are used turns out to be quite complicated.
Perhaps the busreset can be implemented in i2c.cpp – fairly easy as we already know what pins are declared. I would suggest to add this to all platforms as it is a very handy tool to unlock the i2c bus - easier when we create test code – in most cases, we will no longer need to power cycle the board upon a bus lockup. Perhaps we even need to add a call to busreset when the ctor is instantiated to prevent i2c bus lockups at power up. 
Bit banging the pins can be solved using DigitalInOut (code below is not yet tested but should work):
```C
void I2C_busreset(void)
{
    DigitalInOut sclpin(scl, PIN_INPUT, PullNone, 1);
    DigitalIn sdapin(sda);
    if(sclpin == 0 || sdapin == 0)  // I2C bus lock-up occured when SCL/SDA are not high
    {
        while(sclpin == 0)          // bit bang SCL until the offending device releases the bus
        {
            sclpin.output();
            sclpin = 1;
            wait_ms(1);
            sclpin = 0;
            wait_ms(1);
            sclpin.input();
        }
    }
}
```
 
__Notes__
* The 1ms timing is probably a little too slow. Perhaps reading the frequency set by the user or using the default 100KHz frequency would be better.
* Is it safe to assume the scl and sda pins return to their initial state once the busreset function is exited or do we need to add a destructor at the end?
* Above code is prone to hanging the system if scl does not return to a high state. According to most datasheets, bit banging it 10 times should be sufficient, so perhaps we should follow this path and have the function return succes or failure – this way, the user can decide in his code whether he wants to give it another try or not.

0xc0170, 2017-10-12:
There was a patch recently, that will be reopned, for tracking, I'll reference it here: https://github.com/ARMmbed/mbed-os/pull/4838

deepikabhavnani, 2018-10-08:
Fixed by #8109

closed 2018-10-08 by deepikabhavnani