ARMmbed/mbed-os #889: mbed rtos issue with nucleo F401re

araad, 2015-02-06:
I've just started playing around with an ST Nucleo F401REe with the mbed libraries. I've exported the mbed and mbed-rtos libraries to my desktop. I'm using Visual Studio with VisualGDB using the arm-eabi 4.9.2 toolchain.

Everything compiles fine and I'm able to build and upload the simple thread example found on the RTOS handbook page.

The problem that I'm having is when I'm running the example, the call to Thread::wait() in the main thread reaches the Infinite Loop subroutine in startup_stm32f401xe.s

I'm guessing this is not supposed to happen since the comments before Infinite Loop states that I've encountered an unexpected interrupt.

If i remove the call to Thread::wait() from the main thread, the second thread is never being called.

I have the latest librairies, that include the fix mentioned in #775 which seems to be somewhat related to my problem.

With the help of this page: http://www.keil.com/pack/doc/cmsis_rtx/_timer_tick.html
I've noticed that OS_SYSTICK is not defined anywhere, so I've added it myself in order to use the Cortex-M SysTick timer in case it was expecting an alternative timer. I've checked that OS_CLOCK and SystemCoreClock are the same but still doesnt work.

Any suggestions on what's going on?

ohagendorf, 2015-02-11:
I checked, using the current master from here, the RTOS under NUCLEO_F401RE and everyting is as it should be.

All tests
```
singletest.py -M ..\muts_all.json -i ..\test_spec.nucleo_f401.json -n RTOS_x
```
with x=1..8 are OK.

The exports of RTOS_1 to CoIDE and GCC_ARM are running.

You describe a behaviour as we had in #775. Did you export from online compiler or did you clone this repository?
Please compare your startup_stm32f401xe.s with the current one from here. You should see a 
```
bl _start
```
and not a 
```
bl main
```

I'm using gcc 4.9 2014q4 from https://launchpad.net/gcc-arm-embedded and CoIDE 1.7.8.

adamgreen, 2015-02-12:
@araad Have you tried pointing Visual Studio to use the makefile that was placed in the ZIP archive when you exported from the online compiler?  There is a lot of GCC and mbed knowledge encapsulated in those exported makefiles.  If you are using a makefile generated by Visual GDB's wizard then you are losing that knowledge.

0xc0170, 2015-02-26:
@araad any progress with this issue ?

araad, 2015-02-27:
Hi, sorry for the late reply.

I had to do a few things to get it to work. First i switched from VisualGDB's default gcc toolchain to arm-none-eabi. Then i combined some of the GCC and mbed knowledge found in mbed's makefile as pointed out by @adamgreen. That finally solved it for me.

Thanks.

closed 2015-02-27 by araad