ARMmbed/mbed-os #890: Public method return values

neilt6, 2015-02-06:
Just curious, but is there any particular reason that a lot of the public class methods indicate success/failure using an `int` instead of a `bool`?

A few examples:

```c++
//Returns 0 on success (ACK), non-0 on failure (NACK)
int I2C::read(int address, char *data, int length, bool repeated = false);

//Returns 0 on success (ACK), non-0 on failure (NACK)
int I2C::write(int address, const char *data, int length, bool repeated = false);

//Returns 1 if data is available, 0 otherwise
int SPISlave::receive(void);

//Returns 1 if data is available, 0 otherwise
int SerialBase::readable();

//Returns 1 if data can be written, 0 otherwise
int SerialBase::writeable();
```

I realize this convention is similar to a lot of _C Standard Library_ functions, but I for one find it inconsistent and counter-intuitive. For example, while writing to a serial port may look like this:

```c++
//Send an 'A' if possible
if (serial.writeable()) {
    serial.putc('A');
}
```

Checking for success/failure when writing to an I2C device looks like this:

```c++
//Try to send 8B of data to I2C device 0xFF
if (i2c.write(0xFF, data, 8)) {
    //Failure!
} else {
    //Success!
}
```

...or even worse, like this:

```c++
//Try to send 8B of data to I2C device 0xFF
if (!i2c.write(0xFF, data, 8)) {
    //Success!
} else {
    //Failure!
}
```

To me, neither of these statements read intuitively. I first came to mbed from the C# world, and in C# you can't evaluate integers as true or false in an if statement. So I spent a lot of time digging through _mbed-src_, and researching C++ language rules to try to figure out what the integer return values meant. Then, to make matters worse, you have this:

```c++
//Returns 1 on success (ACK), 0 on failure (NACK)
int I2C::write(int data);
```

This function indicates success/failure using the opposite return values! To be fair, this one is meant to be used with the `I2C::RxStatus enum`, but then why not make the return value an `enum` instead of an `int`? The following would have been a much more intuitive API:

```c++
//Returns 'true' on success (ACK), 'false' on failure (NACK)
bool I2C::read(int address, char *data, int length, bool repeated = false);

//Returns 'true' on success (ACK), 'false' on failure (NACK)
bool I2C::write(int address, const char *data, int length, bool repeated = false);

//Returns an RxStatus enum to indicate ACK or NACK
I2C::RxStatus I2C::write(int data);

//Returns 'true' if data is available, 'false' otherwise
bool SPISlave::receive(void);

//Returns 'true' if data is available, 'false' otherwise
bool SerialBase::readable();

//Returns 'true' if data can be written, 'false' otherwise
bool SerialBase::writeable();
```

mbednotifications, 2015-02-06:
Hi Neil, I agree that the return values are not handled consistently and
that enums would be easier to understand and remember. However in the case
of I2C and other more complex interfaces a simple bool does not provide a
lot of info and thus recovery options. An I2C blockwrite has one defined
success (address and all data acked), but many possible failure modes: no
ack on slave address, no ack on one of the databytes, no access to bus since
it was busy or stuck, timeout on access, arbitration lost to another master
when sending address.

So in general an int (or enum) would be best way to get at least some info
on what is going on. However, current libs don't seem to return error types
in a documented way.

 

Regards,

Wim

 

  _____  

From: mbed-devel@googlegroups.com [mailto:mbed-devel@googlegroups.com] On
Behalf Of Neil Thiessen
Sent: vrijdag 6 februari 2015 17:57
To: mbedmicro/mbed
Subject: [mbed] Public method return values (#890)

 

Just curious, but is there any particular reason that a lot of the public
class methods indicate success/failure using an int instead of a bool?

A few examples:

//Returns 0 on success (ACK), non-0 on failure (NACK)
int I2C::read(int address, char *data, int length, bool repeated = false);
 
//Returns 0 on success (ACK), non-0 on failure (NACK)
int I2C::write(int address, const char *data, int length, bool repeated =
false);
 
//Returns 1 if data is available, 0 otherwise
int SPISlave::receive(void);
 
//Returns 1 if data is available, 0 otherwise
int SerialBase::readable();
 
//Returns 1 if data can be written, 0 otherwise
int SerialBase::writeable();

I realize this convention is similar to a lot of C Standard Library
functions, but I for one find it inconsistent and counter-intuitive. For
example, while writing to a serial port may look like this:

//Send an 'A' if possible
if (serial.writeable()) {
    serial.putc('A');
}

Checking for success/failure when writing to an I2C device looks like this:

//Try to send 8B of data to I2C device 0xFF
if (i2c.write(0xFF, data, 8)) {
    //Failure!
} else {
    //Success!
}

...or even worse, like this:

//Try to send 8B of data to I2C device 0xFF
if (!i2c.write(0xFF, data, 8)) {
    //Success!
} else {
    //Failure!
}

To me, neither of these statements read intuitively. I first came to mbed
from the C# world, and in C# you can't evaluate integers as true or false in
an if statement. So I spent a lot of time digging through mbed-src, and
researching C++ language rules to try to figure out what the integer return
values meant. Then, to make matters worse, you have this:

//Returns 1 on success (ACK), 0 on failure (NACK)
int I2C::write(int data);

This function indicates success/failure using the opposite return values! To
be fair, this one is meant to be used with the I2C::RxStatus enum, but then
why not make the return value an enum instead of an int? The following would
have been a much more intuitive API:

//Returns 'true' on success (ACK), 'false' on failure (NACK)
bool I2C::read(int address, char *data, int length, bool repeated = false);
 
//Returns 'true' on success (ACK), 'false' on failure (NACK)
bool I2C::write(int address, const char *data, int length, bool repeated =
false);
 
//Returns an RxStatus enum to indicate ACK or NACK
I2C::RxStatus I2C::write(int data);
 
//Returns 'true' if data is available, 'false' otherwise
bool SPISlave::receive(void);
 
//Returns 'true' if data is available, 'false' otherwise
bool SerialBase::readable();
 
//Returns 'true' if data can be written, 'false' otherwise
bool SerialBase::writeable();

-
<https://github.com/mbedmicro/mbed/issues/890> .
<https://github.com/notifications/beacon/AFAC1yQuD-vrNtzB1vHFbVJkwMuZQBNHks5
npOnwgaJpZM4Dc5OI.gif> 

You received this message because you are subscribed to the Google Groups
"mbed-devel" group.
To unsubscribe from this group and stop receiving emails from it, send an
email to mbed-devel+unsubscribe@googlegroups.com.
For more options, visit https://groups.google.com/d/optout.

neilt6, 2015-02-06:
Hi Wim. That's true, in case of multiple failure modes an `enum` should definitely be returned. However, the I2C class doesn't currently have multiple failure modes, it only indicates if the correct number of bytes were transferred or not. There are also several places where 1's and 0's are simply used in place of true and false (for example all of the `readable()` and `writeable()` methods). In my opinion these should be changed to `bool` instead of `int`.

closed 2017-10-27 by ChiefBureaucraticOfficer