ARMmbed/mbed-os #898: Regarding Ticker event queue processing

masaohamanaka, 2015-02-10:
I think with the following code does not work as expected, if two Tickers are running.

us_ticker_api.c
us_ticker_insert_event()

    /* disable interrupts for the duration of the function */
    ticker_event_t *prev = NULL, *p = head;
    while (p != NULL) {
        /* check if we come before p */
        if ((int)(timestamp - p->timestamp) < 0) {
            break;
        }
        /* go to the next element */
        prev = p;
        p = p->next;
    }

If the current time value was 0xFFFFFF00.

For example, Ticker1 calls us_ticker_insert_event() with "timestamp = 0x00000020", head-> timestamp will be 0x00000020.
After this, if Ticker2 (short period) calls us_ticker_insert_event() with "timestamp = 0xFFFFFFE0",
The correct order is that interrupt occurs in 1, 2 as below.

1. 0xFFFFFFE0 (for Ticerk 2) <-not ignored

---time wrap arround---------
2. 0x00000020 (for Ticerk 1)

But, since the code above only compare the simple head-> timestamp and timestamp, comparison result will be as below.

0xFFFFFFE0(for Ticerk 2) - 0x00000020(for Ticerk 1) > 0

As a result, the interrupt is generated in the following order .

-. 0xFFFFFFE0 (for Ticerk 2) <- ignored

---time wrap arround---------
1. 0x00000020 (for Ticerk 1)
.......
2. 0xFFFFFFE0 (for Ticerk 2)

Sissors, 2015-02-10:
It should go correctly since it is explicitly set as 32-bit integer (actually it isn't, would be nicer if it would say int32_t, but this does the same for the mbed platform).

That is because 0xFFFFFFE0 - 0x000000020 = 0xFFFFFFC0, which is as 32-bit signed integer equal to -64, which is small than zero.

masaohamanaka, 2015-02-13:
Thanks your comment.

I think my example was not good.
How about below case?

1) If the current time value was 0x70000000.
2) Ticker1 calls us_ticker_insert_event() with "timestamp = 0x00000020"
3) Ticker2 calls us_ticker_insert_event() with "timestamp = 0x7FFFFFE0"

The result is 0x7FFFFFE0 - 0x000000020 = 0x7FFFFFC0 > 0.

But, I believe any user do not wait about 30 minute...

Sissors, 2015-02-13:
Since the ticker uses 32-bit signed integer, it will assume everything further away than roughly 30 minutes happened in the past. So you are correct that in your example it will go wrong, but this is because in general it is now allowed to have a ticker period larger than 0x7FFFFFFF.

sg-, 2017-01-10:
Will keep this open as a request make tickers a 64bit timestamp

closed 2017-10-27 by ChiefBureaucraticOfficer