ARMmbed/mbed-os #902: nrf51xx RTOS issue

0xc0170, 2015-02-11:
It was added a while ago, #337. There has been several reports that is not functional, in the online IDE or exported to GCC. Some links to reports about this issue on mbed.org are here https://developer.mbed.org/questions/6235/Which-RTOS-did-you-successfully-run-on-n/?c=15726

Looks to me that the problem is with scheduling. Simple application like main() with Thread::wait() wont do anything.

0xc0170, 2015-02-11:
Looking at it, systick is not active. The documentation states (link http://developer.nordicsemi.com/nRF51_SDK/doc/7.0.1/s110/html/a00080.html):

nRF51 does not have a SysTick timer. Therefore, RTC1 is used instead.

Which would explain why is not functional.

0xc0170, 2015-02-11:
As a result, I am going to send PR to remove nrf51 support for RTX, so it fails to build with a proper error message, that device is not supported.

closed 2015-02-12 by 0xc0170

finneyj, 2015-03-27:
This is becoming a popular chip, and it would be very useful to see threading support for it (which seems to only part of the RTOS library)? Are there any plans to add support the NRF51822 using the alternative timer that you mentioned?

rosterloh, 2015-03-27:
+1 to this

commit e6c32b54 (2015-02-12): Merge pull request #909 from 0xc0170/fix_#902

Fix #902 - revert RTOS support for NRF51822