ARMmbed/mbed-os #910: IAR templates for Kinetis - Simulator as debugger

0xc0170, 2015-02-12:
As I was testing the PR, I used KL25Z to export blinky project. The exported project has predefined Simulator. Once I set it to CMSIS-DAP, the default flash algo are used which are not correct. Received flash write errors and application did not run. I recall flash algo should be overwrittten to Freescale provided in the Freescale folder.

@GustavWi Shall we change the templates to work out of the box?

GustavWi, 2015-02-12:
Hi Martin,
Ok I have not set the debugger to hardware. I thought it would be better if the user can make the appropriate settings themselves. I  would not call this a bug because I actively did this. If it is the standard for exporters to have hardware debugging as default I can change all of them.

0xc0170, 2015-02-12:
It should be "out-of-box" approach for exporters. Please look also at flash algo, I believe some kinetis devices need to override the default one? You might know better :)

GustavWi, 2015-02-12:
Yes I am going to look at the cmsis-dap KL25Z debugging in the EW now.

Thanks I will set the appropriate settings for hardware debugging. There is however one complication. All debugging information is located in a separate file (ewd). I will have to add one of these as well. Expect all this to be done at the end of next week or the week after. I have a lot of other things to do as well.

0xc0170, 2015-02-12:
It's fine. Either me or you will do it. Thanks for interest in this issue

commit 4e403f87 (2015-02-24): Merge pull request #924 from GustavWi/iar_mbed

Fix #910 - debug file with predefined hardware debug settings
closed 2015-02-24 by 0xc0170