ARMmbed/mbed-os #911: ST Nucleo F334R8 - PwmOut not working

samcol02, 2015-02-13:
On ST Nucleo F334R8, the PwmOut class does not drive any of the PWM pins correctly. After setup, the value is always high, regardless of period or pulsewidth.

The example code I was using at the time is linked below:
https://developer.mbed.org/users/gask0032/code/Nucleo_pwm/

The mbed library (revision 93:e188a91d3eaa) and firmware (V2.J23.M6) were all up to date.

0xc0170, 2015-02-26:
@bcostm - Is this known issue?

bcostm, 2015-03-06:
When I have released the F334R8 few months ago, I had no problem with pwm. I'll look at it.

sg-, 2017-01-10:
Can we confirm if this passes using https://github.com/ARMmbed/ci-test-shield

samcol02, 2017-01-11:
Sorry, can't help there, I don't have the device anymore. It sounds like the issue should have been closed after bcostm tested it with the newer revision/firmware. Thanks for following up.

closed 2017-01-11 by samcol02