ARMmbed/mbed-os #913: Nucleo F401RE Target: Ticker BUG - Device freezes after 72 minutes

richardtreu, 2015-02-13:
Used online IDE/compiler (so I assume that this will be the latest mbed version). It is definitely the problem described and commented by me in the following link:

http://developer.mbed.org/questions/5291/Timeouts-stop-working-at-desired-rate-af/

Sissors, 2015-02-13:
It will only be the latest mbed version if you update it manually! (Or create a new program with default settings). This is because you do not want code mysteriously breaking when a new mbed version is released which changes some things.

If you click on your mbed lib you can see which version it is, and at the right side there will be an update button if you don't have the latest one.

richardtreu, 2015-02-13:
This function seems too unobvious to me. Please think about redesigning. I can't remember reading about this in the manual.
My version was from Oct 2014. Now I updated and hope...

closed 2015-02-13 by richardtreu