ARMmbed/mbed-os #916: OS dependent copy call in exporter scripts

BlackstoneEngineering, 2015-02-13:
in the workspace_tools/export/__init__.py file, the export function does this
```python
 if ide is None:
        # Simply copy everything, no project files to be generated
        for d in ['src', 'lib']:
            os.system("cp -r %s/* %s" % (join(project_path, d), tempdir))
        report = {'success': True}
```

that os.system call uses cp -r, which is not OS independent. This should get fixed or its going to bite us.

BlackstoneEngineering, 2015-02-13:
@screamerbg @sg- , tagging y'all to make you aware of it.

BlackstoneEngineering, 2015-02-13:
@0xc0170 I assigned it to you because you fix things, like alot of things, and i figured if anyone would get around to it it would be you, feel free to re-assign as needed.

0xc0170, 2015-02-15:
It was replaced ,2 days ago, probably once you added this? @screamerbg pushed to master, did not reference this, so I can't answer if it was fix or not.

screamerbg, 2015-02-19:
@0xc0170 yes, this was fixed and deployed on developer.mbed.org.

closed 2015-02-19 by screamerbg