ARMmbed/mbed-os #920: USART2 issue on STM32F411RE Nucleo

joetijerina, 2015-02-19:
There seems to be a problem with USART2 on the STM32F411 Nucleo board.

When streaming a file through the device (with a PC on both
sides) to exercise the serial ports (simple passthrough application using the MBED Serial class)

    // USART2
    Serial usb(USBTX, USBRX);
    // USART1
    Serial ext(D8, D2);
    usb.baud(<some baud>);
    ext.baud(<some baud>);
    while (true) {
    if (usb.readable())
        ext.putc(usb.getc());
    if (ext.readable())
        usb.putc(ext.getc());
    }

Test was against revision 06e2b3c8802cb4f78e9958ba9923755bd458e8b7
of the master branch of https://github.com/mbedmicro/mbed, building with both the GCC ARM Embedded and IAR toolchains.  Also tested against code built against revision 93 of the MBED library in the online compiler.  The issue usually manifests itself when streaming data into USART1 or USART6 and out USART2.  The file being streamed through is ASCII text and is ~100kB in size.

At low baud rates (less than 57600), all the data seems to make it through.  At higher baud rates, however, only the first 1 - 5 kB of data is seen on the other side and have to reset the Nucleo board before you can get more data through.  

Can you look into this issue and let me know if you also notice the same problem and provide a solution? If a bug is found in the driver stack, please update the mbed library and notify me.

Thanks

sg-, 2017-01-10:
I cant comment on the specific F411 problem needing to be reset but I've seen something similar and has to do with Serial inheriting from Stream and retargeting though the standard library where RawSerial would work upto about 921600 baud when bypassing the standard library hooks.

https://developer.mbed.org/users/sam_grove/code/data_loss_test/

LMESTM, 2017-05-02:
@joetijerina @sg-  This looks to me like a duplicate of #687 - isn't it ?  I'd propose to close as the issue was discussed in #687

closed 2017-05-03 by sg-