ARMmbed/mbed-os #942: LPC824 Serial receive not registering  chars sent.

GustavWi, 2015-03-02:
Hi I ran the example program serial echo (A9) on LPCXpresso824 -MAX. I tried it with GCC_ARM, uARM and IAR. The program is simply stuck in ```while (!serial_readable(obj));```.

```
int serial_getc(serial_t *obj)
{
    while (!serial_readable(obj));
    return obj->uart->RXDAT;
}
```

toyowata, 2015-03-02:
Hi,

Did you read a warning message in the platform page?

https://developer.mbed.org/platforms/LPCXpresso824-MAX/

>Warning
>
>By default, the USBRX pin doesn't connect USB virtual serial of the interface chip. You must change the position of 0-ohm register SJ9 in between 2 (TARGET_RX-P0_18) and 3 (LINK_TX). Please refer the schematic for more detail.

You have to change SJ9 0-ohm register setting if you need serial input using this platform.

GustavWi, 2015-03-02:
Nope I missed it. I soldered the connection and I can see the USBRX responding when I send data on the oscilloscope. It places the received data in TXDATA but it is never transferred back. Have you tested this and had a correct result?

toyowata, 2015-03-03:
>Have you tested this and had a correct result? 

Yes.  I tested with an example code in the Serial API.
https://developer.mbed.org/handbook/Serial#hello-world

And also, I tested STDIO automated-test which has both serial TX/RX functions.
Can you send me your test case?

0xc0170, 2015-03-03:
@toyowata A9 - the test program in the mbed SDK tests

GustavWi, 2015-03-03:
I tested your program and it works. I did some more testing and found this:
```
#include "mbed.h"
#include "test_env.h"

#define TXPIN     USBTX
#define RXPIN     USBRX

namespace {
    const int BUFFER_SIZE = 48;
    char buffer[BUFFER_SIZE] = {0};
}

int main() {
    MBED_HOSTTEST_TIMEOUT(20);
    MBED_HOSTTEST_SELECT(echo);
    MBED_HOSTTEST_DESCRIPTION(Serial Echo at 115200);
    MBED_HOSTTEST_START("MBED_A9");

    Serial pc(TXPIN, RXPIN);
    pc.baud(115200);

    while (1) {
        pc.gets(buffer, BUFFER_SIZE - 1);
        pc.printf("%s", buffer);
    }
}
```
If you comment out the MBED_HOSTTEST... calls the program works. Serial pc uses USART0 when the MBED_HOSTTEST... calls are commented out otherwise USART1. It successfully reads the data in both configurations but fails to transmit the data.

toyowata, 2015-03-03:
OK.  This is a defect in LPC824 HAL of Serial API.
get_available_uart() should check existing resource with PinName setting to avoid creating other UART config.

toyowata, 2015-04-07:
The fix has been merged.
https://github.com/mbedmicro/mbed/pull/1009

closed 2015-04-07 by toyowata