ARMmbed/mbed-os #955: ST Nucleo 411RE wait() issue in offline export

BlackstoneEngineering, 2015-03-06:
# Problem
the wait() function does not execute properly on the ST Nucleo F411RE board when exported to keil, the wait_ms() function behaves as expected. I believe the function is improperly defined in the port.

for example
```C
#include "mbed.h"

DigitalOut myled(LED1);
DigitalOut toggle(D7);

int main()
{
    while(1) {
        myled = 0; // LED is OFF
        toggle = 0;
        wait(1);
        myled = 1; // LED is ON
        toggle = 1;
        wait(1); 
    }
}
```
### Online Compiler OK
The above code will behave as expected when compiled online, the led will blink and the toggle will toggle, with 1s intervals.
[image]

### Offline Compiler problem (keil)
 but when exported to keil the wait function is not behaving properly, the timings are not executing properly with ~2us intervals, which leads me to believe the wait function is entering and immediately exiting. 
[image]

### hotfix
by using wait_us() in keil we can get the expected behavior.

# Acceptance Criteria
the wait() function behaves as expected

@screamerbg

Sissors, 2015-03-06:
Previous problems regarding wait while wait_ms/wait_us works properly were always related to the FPU not being set properly. Although this generally resulted in a freeze iirc. Still something which should be looked at.

dbestm, 2015-03-09:
Could you tell me how is precompiled the mbed Library?

I can reproduce the problem only with the mbed precompiled library when I export it.
When I use the mbed source instead, there is no problem even when I export the project.
I try to set the HW FPU, but there is no impact on the behavior of the blink.

dinau, 2015-03-26:
The *.tmpl file has wrong device name (STM32F401RC),
https://github.com/mbedmicro/mbed/blob/master/workspace_tools/export/uvision4_nucleo_f411re.uvopt.tmpl
I've changed the device name on project option from STM32F401RC to
STM32F411RE.
"wait(1)" function works well.

0xc0170, 2015-03-26:
@dinau will you send PR to address this?

dinau, 2015-03-27:
I've tired the exported project files on "uVision5 IDE" and
changed the device from F401RE to F411RE.
"wait()" function works well above mentioned.
But "uVision4 IDE" dose not support the device file for F411RE
at this moment.
Therefore it seems that anyone could not use the exported project files
 with F411RE on "uVision4 IDE".

bcostm, 2016-05-10:
To be closed

dinau, 2016-05-10:
Is this issue resolved ?

dinau

0xc0170, 2016-05-10:
@bcostm I closed few issues I considered are fixed, please can you review and provide references to the commits which should have fixed them.

bcostm, 2017-04-07:
I just generated a uVision Keil project from the mbed IDE on the NUCLEO_F411RE and NUCLEO_F401RE using the latest mbed-dev library.
I confirm that the wait function is working well. I have tried different values below and above one second.
So, I think it can be closed.

TO_BE_CLOSED

dinau, 2017-04-09:
As uVision4 IDE in online export menu has been deleted,
it's no problem by necessity.
I agree to be closed.

closed 2017-04-10 by 0xc0170