ARMmbed/mbed-os #968: FRDM_K64F, SPI MODE 3, MOSI error

z1z2, 2015-03-13:
In the following example, I supposed to send 170 ( 0xAA), but received 171. It seems that the MOSI jumps up too early and makes the LSB always be "1". How can I solve the problem?

[image]

```C
#include "mbed.h"
 
SPI device(PTD6,NC,PTD5);
 
int main() {
    int i = 0;
    device.format(8,3);
    while(1) {
        device.write(0x55);
        device.write(0xAA);
        wait(1);
    }
}
```

0xc0170, 2015-03-13:
Is this only with format(8, 3), can you describe a bit more.. There's test in the mbed , let me share it with you here, if we can run this to fail: https://github.com/mbedmicro/mbed/blob/master/libraries/tests/mbed/spi_master/main.cpp

Sissors, 2015-03-13:
I think this might be related to this one: https://github.com/mbedmicro/mbed/issues/840

z1z2, 2015-03-14:
@0xc0170 , I can not build the test in the online compiler, But I tested the other 3 modes. Mode 0 and 2 are OK, but mode 1 failed, too.
[image]

Sissors, 2015-05-05:
This bug should be solved with the pull request above

closed 2015-05-22 by 0xc0170