ARMmbed/mbed-os #970: NO PORT_I in the file ? mbed/libraries/mbed/targets/hal/TARGET_STM/TARGET_STM32F4XX/PinNames.h

idodoyo, 2015-03-14:
NO PORT_I??

typedef enum {
    // STM32 Pin Names
    PA_0 = 0, PA_1, PA_2, PA_3, PA_4, PA_5, PA_6, PA_7, PA_8, PA_9, PA_10, PA_11, PA_12, PA_13, PA_14, PA_15,
    PB_0, PB_1, PB_2, PB_3, PB_4, PB_5, PB_6, PB_7, PB_8, PB_9, PB_10, PB_11, PB_12, PB_13, PB_14, PB_15,
    PC_0, PC_1, PC_2, PC_3, PC_4, PC_5, PC_6, PC_7, PC_8, PC_9, PC_10, PC_11, PC_12, PC_13, PC_14, PC_15,
    PD_0, PD_1, PD_2, PD_3, PD_4, PD_5, PD_6, PD_7, PD_8, PD_9, PD_10, PD_11, PD_12, PD_13, PD_14, PD_15,
    PE_0, PE_1, PE_2, PE_3, PE_4, PE_5, PE_6, PE_7, PE_8, PE_9, PE_10, PE_11, PE_12, PE_13, PE_14, PE_15,
    PF_0, PF_1, PF_2, PF_3, PF_4, PF_5, PF_6, PF_7, PF_8, PF_9, PF_10, PF_11, PF_12, PF_13, PF_14, PF_15,
    PH_0, PH_1, PH_2, PH_3, PH_4, PH_5, PH_6, PH_7, PH_8, PH_9, PH_10, PH_11,

    LED1        = PD_13,
    LED2        = PD_12,
    LED3        = PD_13,
    LED4        = PD_12,
    LED5        = PD_14,
    LED6        = PD_15,

    // Not connected
    NC = (int)0xFFFFFFFF
} PinName;

0xc0170, 2015-03-16:
Port names are in the different header, in the PortName

```
typedef enum {
    PortA = 0,
    PortB = 1,
    PortC = 2,
    PortD = 3,
    PortE = 4,
    PortF = 5,
    PortG = 6,
    PortH = 7,
    PortI = 8
} PortName;
```

What are you reporting?

idodoyo, 2015-03-17:
which file？

0xc0170, 2015-03-17:
https://github.com/mbedmicro/mbed/blob/master/libraries/mbed/targets/hal/TARGET_STM/TARGET_STM32F4XX/PortNames.h

Please provide more details when you are reporting an issue.

idodoyo, 2015-03-17:
ok,i find the file, thanks, i just want use PORT PIN derictly.

closed 2017-01-10 by sg-