ARMmbed/mbed-os #993: mbed test script "vtor_reloc" is missing alignment

RobMeades, 2015-03-23:
I've just been trying to get the `vtor_reloc` script to run and have spotted that the vector table is missing the alignment attribute.  The line:

`uint32_t int_table[NUM_VECTORS];`

...should read:

`uint32_t int_table[NUM_VECTORS] __attribute__ ((aligned(256)));`

...for the script to work reliably.  It must have been working by luck up to now.

RobMeades, 2015-03-23:
I've created pull request:

https://github.com/mbedmicro/mbed/pull/994

...to cover this.

PrzemekWirkus, 2015-03-30:
Hello Rob,
this test script was failing, thank you for fixing this issue.
We will merge it soon I hope! :)

PrzemekWirkus, 2015-05-02:
Fix is already in the code so we can close this issue: https://github.com/mbedmicro/mbed/blob/master/libraries/tests/mbed/vtor_reloc/main.cpp

closed 2015-05-02 by PrzemekWirkus