ARMmbed/mbed-os #996: Export to IAR failing

mfiore02, 2015-03-24:
Revision ea01d61fa18430564b78226045b196bb6bf6b66a (Tue Mar 24 08:48:37 2015 +0000) of master branch.

mfiore@mwf-ubuntu:~/work/mbed_devel/mbed$ python workspace_tools/project.py -c -m NUCLEO_F411RE -i iar -n MBED_BLINKY
Copy: test_env.h
Copy: gpio_object.h
Copy: PeripheralPins.h
Copy: PinNames.h
Copy: objects.h
Copy: device.h
Copy: PortNames.h
Copy: PeripheralNames.h
Copy: core_cm4.h
Copy: core_cm4_simd.h
Copy: core_cm3.h
Copy: core_cm0plus.h
Copy: core_ca_mmu.h
Copy: core_caInstr.h
Copy: core_caFunc.h
Copy: core_ca9.h
Copy: core_cmFunc.h
Copy: core_cmInstr.h
Copy: core_cm0.h
Copy: stm32f4xx_hal_sdram.h
Copy: stm32f4xx_hal_rcc.h
Copy: stm32f4xx_hal_can.h
Copy: stm32f4xx_ll_usb.h
Copy: stm32f4xx_hal_dma.h
Copy: stm32f4xx_hal_rng.h
Copy: stm32f4xx_hal_flash_ex.h
Copy: stm32f4xx_hal_dcmi.h
Copy: stm32f4xx_hal_pccard.h
Copy: stm32f4xx_hal_dac_ex.h
Copy: stm32f4xx_hal_cryp_ex.h
Copy: stm32f4xx_hal_usart.h
Copy: stm32f4xx_hal_gpio_ex.h
Copy: stm32f4xx_hal_dma_ex.h
Copy: stm32f4xx_hal_smartcard.h
Copy: stm32f4xx_hal_eth.h
Copy: stm32f4xx_hal_i2s.h
Copy: stm32f4xx_hal_gpio.h
Copy: stm32f4xx_hal_sai.h
Copy: stm32f4xx_hal_dac.h
Copy: stm32f4xx_ll_sdmmc.h
Copy: stm32f4xx_hal_dma2d.h
Copy: stm32f4xx_hal_flash.h
Copy: stm32f4xx_hal_ltdc.h
Copy: stm32f4xx_hal_sram.h
Copy: stm32f4xx_hal_pwr_ex.h
Copy: stm32f4xx_hal_irda.h
Copy: stm32f4xx_hal_conf.h
Copy: stm32f4xx_hal_cryp.h
Copy: stm32f4xx_hal_tim_ex.h
Copy: stm32f4xx_ll_fsmc.h
Copy: stm32f4xx_hal_def.h
Copy: stm32f4xx_hal.h
Copy: stm32f4xx_hal_pcd.h
Copy: stm32f4xx_hal_adc_ex.h
Copy: stm32f4xx_ll_fmc.h
Copy: stm32f4xx_hal_i2s_ex.h
Copy: stm32f4xx_hal_hash_ex.h
Copy: stm32f4xx_hal_wwdg.h
Copy: stm32f4xx_hal_crc.h
Copy: stm32f4xx_hal_sd.h
Copy: stm32f4xx_hal_nor.h
Copy: stm32f4xx_hal_pcd_ex.h
Copy: stm32f4xx_hal_hash.h
Copy: stm32f4xx_hal_i2c.h
Copy: stm32f4xx_hal_hcd.h
Copy: stm32f4xx_hal_uart.h
Copy: stm32f4xx_hal_pwr.h
Copy: stm32f4xx_hal_flash_ramfunc.h
Copy: stm32f4xx_hal_cortex.h
Copy: stm32f4xx_hal_spi.h
Copy: stm32f4xx_hal_iwdg.h
Copy: stm32f4xx_hal_i2c_ex.h
Copy: stm32f4xx_hal_nand.h
Copy: stm32f4xx_hal_rcc_ex.h
Copy: stm32f4xx_hal_tim.h
Copy: stm32f4xx_hal_rtc_ex.h
Copy: stm32f4xx_hal_rtc.h
Copy: stm32f4xx_hal_adc.h
Copy: system_stm32f4xx.h
Copy: cmsis.h
Copy: stm32f4xx.h
Copy: stm32f411xe.h
Copy: cmsis_nvic.h
Copy: hal_tick.h
Copy: us_ticker_api.h
Copy: rtc_api.h
Copy: i2c_api.h
Copy: pinmap.h
Copy: spi_api.h
Copy: can_api.h
Copy: gpio_irq_api.h
Copy: analogout_api.h
Copy: sleep_api.h
Copy: gpio_api.h
Copy: pwmout_api.h
Copy: serial_api.h
Copy: analogin_api.h
Copy: port_api.h
Copy: ethernet_api.h
Copy: mbed_error.h
Copy: RawSerial.h
Copy: FileSystemLike.h
Copy: platform.h
Copy: InterruptIn.h
Copy: TimerEvent.h
Copy: semihost_api.h
Copy: DigitalIn.h
Copy: I2C.h
Copy: DirHandle.h
Copy: rtc_time.h
Copy: SPISlave.h
Copy: can_helper.h
Copy: toolchain.h
Copy: BusOut.h
Copy: DigitalOut.h
Copy: I2CSlave.h
Copy: Timeout.h
Copy: SPI.h
Copy: AnalogOut.h
Copy: InterruptManager.h
Copy: mbed.h
Copy: SerialBase.h
Copy: PortIn.h
Copy: AnalogIn.h
Copy: FunctionPointer.h
Copy: PortInOut.h
Copy: wait_api.h
Copy: BusIn.h
Copy: PwmOut.h
Copy: LocalFileSystem.h
Copy: Serial.h
Copy: FilePath.h
Copy: BusInOut.h
Copy: DigitalInOut.h
Copy: FileHandle.h
Copy: mbed_debug.h
Copy: mbed_assert.h
Copy: mbed_interface.h
Copy: PortOut.h
Copy: Ethernet.h
Copy: Timer.h
Copy: FileBase.h
Copy: Ticker.h
Copy: CAN.h
Copy: FileLike.h
Copy: Stream.h
Copy: CallChain.h
Copy: cmain.s
Copy: startup_stm32f411xe.s
Copy: port_api.c
Copy: rtc_api.c
Copy: spi_api.c
Copy: mbed_overrides.c
Copy: analogout_api.c
Copy: serial_api.c
Copy: pwmout_api.c
Copy: gpio_api.c
Copy: analogin_api.c
Copy: us_ticker.c
Copy: i2c_api.c
Copy: pinmap.c
Copy: sleep.c
Copy: gpio_irq_api.c
Copy: PeripheralPins.c
Copy: stm32f4xx_hal_dma2d.c
Copy: stm32f4xx_hal_i2s.c
Copy: stm32f4xx_hal_i2c_ex.c
Copy: stm32f4xx_hal_gpio.c
Copy: stm32f4xx_hal_iwdg.c
Copy: stm32f4xx_hal_rcc_ex.c
Copy: stm32f4xx_hal.c
Copy: stm32f4xx_hal_eth.c
Copy: stm32f4xx_hal_can.c
Copy: stm32f4xx_hal_sd.c
Copy: stm32f4xx_hal_pwr.c
Copy: stm32f4xx_ll_usb.c
Copy: stm32f4xx_hal_rtc.c
Copy: stm32f4xx_hal_i2s_ex.c
Copy: stm32f4xx_hal_rtc_ex.c
Copy: stm32f4xx_hal_dcmi.c
Copy: stm32f4xx_hal_tim_ex.c
Copy: stm32f4xx_hal_adc.c
Copy: stm32f4xx_hal_i2c.c
Copy: stm32f4xx_hal_dma.c
Copy: stm32f4xx_hal_flash_ramfunc.c
Copy: stm32f4xx_hal_hash_ex.c
Copy: stm32f4xx_hal_dma_ex.c
Copy: stm32f4xx_hal_smartcard.c
Copy: stm32f4xx_hal_pcd_ex.c
Copy: stm32f4xx_ll_fsmc.c
Copy: stm32f4xx_hal_spi.c
Copy: stm32f4xx_hal_nor.c
Copy: stm32f4xx_hal_tim.c
Copy: stm32f4xx_ll_sdmmc.c
Copy: stm32f4xx_hal_dac.c
Copy: stm32f4xx_hal_pcd.c
Copy: stm32f4xx_hal_rcc.c
Copy: stm32f4xx_hal_cryp_ex.c
Copy: stm32f4xx_hal_dac_ex.c
Copy: stm32f4xx_hal_hash.c
Copy: stm32f4xx_hal_usart.c
Copy: stm32f4xx_hal_rng.c
Copy: stm32f4xx_ll_fmc.c
Copy: stm32f4xx_hal_adc_ex.c
Copy: stm32f4xx_hal_pccard.c
Copy: stm32f4xx_hal_uart.c
Copy: stm32f4xx_hal_flash_ex.c
Copy: stm32f4xx_hal_hcd.c
Copy: stm32f4xx_hal_pwr_ex.c
Copy: stm32f4xx_hal_sram.c
Copy: stm32f4xx_hal_sai.c
Copy: stm32f4xx_hal_irda.c
Copy: stm32f4xx_hal_flash.c
Copy: stm32f4xx_hal_cryp.c
Copy: stm32f4xx_hal_cortex.c
Copy: stm32f4xx_hal_sdram.c
Copy: stm32f4xx_hal_ltdc.c
Copy: stm32f4xx_hal_crc.c
Copy: stm32f4xx_hal_nand.c
Copy: stm32f4xx_hal_wwdg.c
Copy: system_stm32f4xx.c
Copy: cmsis_nvic.c
Copy: hal_tick.c
Copy: gpio.c
Copy: pinmap_common.c
Copy: us_ticker_api.c
Copy: wait_api.c
Copy: assert.c
Copy: mbed_interface.c
Copy: error.c
Copy: semihost_api.c
Copy: rtc_time.c
Copy: board.c
Copy: test_env.cpp
Copy: Timer.cpp
Copy: FileSystemLike.cpp
Copy: I2C.cpp
Copy: SPISlave.cpp
Copy: SerialBase.cpp
Copy: BusIn.cpp
Copy: RawSerial.cpp
Copy: FileBase.cpp
Copy: CallChain.cpp
Copy: retarget.cpp
Copy: I2CSlave.cpp
Copy: InterruptManager.cpp
Copy: Timeout.cpp
Copy: LocalFileSystem.cpp
Copy: Ethernet.cpp
Copy: Stream.cpp
Copy: FunctionPointer.cpp
Copy: Ticker.cpp
Copy: FilePath.cpp
Copy: FileLike.cpp
Copy: SPI.cpp
Copy: BusOut.cpp
Copy: CAN.cpp
Copy: BusInOut.cpp
Copy: Serial.cpp
Copy: TimerEvent.cpp
Copy: InterruptIn.cpp
Copy: stm32f411xe.icf
Copy: main.cpp
Traceback (most recent call last):
  File "workspace_tools/project.py", line 179, in <module>
    tmp_path, report = export(project_dir, test.id, ide, mcu, EXPORT_WORKSPACE, EXPORT_TMP, extra_symbols=lib_symbols)
  File "/home/mfiore/work/mbed_devel/mbed/workspace_tools/export/__init__.py", line 84, in export
    exporter.generate()
  File "/home/mfiore/work/mbed_devel/mbed/workspace_tools/export/iar.py", line 80, in generate
    iar_files.insert_file(source)
  File "/home/mfiore/work/mbed_devel/mbed/workspace_tools/export/iar.py", line 144, in insert_file
    self.sub_folders[folder_name].insert_file(source_input)
  File "/home/mfiore/work/mbed_devel/mbed/workspace_tools/export/iar.py", line 144, in insert_file
    self.sub_folders[folder_name].insert_file(source_input)
  File "/home/mfiore/work/mbed_devel/mbed/workspace_tools/export/iar.py", line 133, in insert_file
    folder_name = re.match( _reg_exp, dir_sources).group(1)
AttributeError: 'NoneType' object has no attribute 'group'

GustavWi, 2015-03-25:
I tested it on my machine, its windows, it works. I will have to try this on a linux machine.

0xc0170, 2015-03-25:
We should look at https://github.com/mbedmicro/mbed/blob/master/workspace_tools/export/iar.py#L128, looks like using windows paths.

mfiore02, 2015-03-25:
Sorry, should have specified that my machine is Linux.  Ubuntu 12.04

GustavWi, 2015-03-31:
I works properly in beta mode now. I don't know when they will make the update normal mode.

mfiore02, 2015-04-13:
I was working with the offline SDK.  Just tried from the latest master and it works now.  Thanks!

closed 2015-04-13 by mfiore02