ARMmbed/mbed-os #1011: LPC1549 Serial.format will disable the serial port

jurgis, 2015-04-05:
If format function of the serial port is called it will disable the serial port.

For example this will not print anything:
```c++
#include "mbed.h"

Serial pc(USBTX, USBRX);

int main()
{
    pc.baud(115200);
    pc.format(8, SerialBase::None, 1);
    pc.printf("Program started\r\n");
    
    for(;;){}
}
```

I will submit a pull request shortly.

jurgis, 2015-04-05:
The pull request: https://github.com/mbedmicro/mbed/pull/1012

alexbeer2048, 2015-07-27:
If I'm not mistaken this issue can be closed, since the fix is already merged to the master. Right?

On a related subject: I had the same same behavior of Serial::format() on the LPC824 as decribed above.
I managed to to fix it by applying the exact same fix to the serial_api.c as for the LPC1549 target from pull request : https://github.com/mbedmicro/mbed/pull/1012

I'm still a bit of a github noob, so I might be posting this at the wrong place. Should I post a new issue for the LPC824 or create a pull request for the fix right away?

0xc0170, 2015-07-27:
@alexbeer2048 Yes please, you can jsut send PR , reference this issue there that it is related to your platform as well.

closed 2015-07-27 by 0xc0170