ARMmbed/mbed-os #1018: [FRDM-K64F] Cannot address ADC1_SE18

DC37, 2015-04-07:
https://github.com/mbedmicro/mbed/blob/master/libraries/mbed/targets/hal/TARGET_Freescale/TARGET_KPSDK_MCUS/TARGET_MCU_K64F/TARGET_FRDM/PinNames.h

The file appears to have the plumbing for one to address the DAC out pin using the AnalogOut interface, but there does not appear to be a way to be a way to address the ADC1_SE18 pin...

0xc0170, 2015-04-07:
This is ADC pins for the board:

```
const PinMap PinMap_ADC[] = {
    {PTC2,  ADC0_SE4b, 0},
    {PTC8,  ADC1_SE4b, 0},
    {PTC9,  ADC1_SE5b, 0},
    {PTD1,  ADC0_SE5b, 0},
    {PTC10, ADC1_SE6b, 0},
    {PTD5,  ADC0_SE6b, 0},
    {PTC11, ADC1_SE7b, 0},
    {PTD6,  ADC0_SE7b, 0},
    {PTB0 , ADC0_SE8 , 0},
    {PTB1 , ADC0_SE9 , 0},
    {PTB2 , ADC0_SE12, 0},
    {PTB3 , ADC0_SE13, 0},
    {PTC0 , ADC0_SE14, 0},
    {PTB10, ADC1_SE14, 0},
    {PTB11, ADC1_SE15, 0},
    {PTC1 , ADC0_SE15, 0},
    {PTA17, ADC1_SE17, 0},
    //{PTE24, ADC0_SE17, 0}, //I2C pull up
    //{PTE25, ADC0_SE18, 0}, //I2C pull up
    {NC   , NC       , 0}
};
```

I don't see there ADC1_SE18 . Is it available? If yes, can you send PR?

0xc0170, 2015-04-07:
I see now, that pin and for example SE23 don't have any real pin number, same as it is for DAC. You can send PR implementing this additional pins.

spsancti, 2015-09-26:
Is there any solution? I even cannot imagine where to find information about mapping pins to addresses. What to google?

DC37, 2015-09-26:
I have not played with this for a while, but I imagine reading up on the datasheet + user manual of the microcontroller is probably the best bet...

spsancti, 2015-09-26:
It's about 800 pages of reference manual and at the first sight I've found nothing about the DAC and 0xFEFE from mbed. Maybe, is there a special term for it?

DC37, 2015-09-26:
Hrm, I don't remember, but what I do remember is attempting to address the pin by playing around (not super rigorously) with the MBed abstractions and failing pretty miserably to get any readings at all. But I haven't gotten a good look at the hardware or how it's even configured - I imagine that it will take a little more effort and register-flipping to get this right.

spsancti, 2015-09-26:
Thank you anyway, I'll try to dig it a little deeper.

spsancti, 2015-09-27:
Can you tell me, whence goes the DAC0_OUT = 0xFEFE? I haven't found anything similar in the RM. 0x4003_F000 is the absolute address for DAC0 and the slot 63 in the AIPS-Lite 0. Maybe 0xFEFE is the second part of addres (0x4003_FEFE)? Where to look for EFE?

0xc0170, 2015-09-27:
@spsancti Please look at https://github.com/mbedmicro/mbed/blob/master/libraries/mbed/targets/hal/TARGET_Freescale/TARGET_KPSDK_MCUS/TARGET_MCU_K64F/TARGET_FRDM/PinNames.h#L240 , there's a comment that the pin does not have a valid number in the reference manual, therefore it was "random". Do you want to add new pins to ADC like ADC1_SE18 ?

spsancti, 2016-08-01:
Cool, but what to do with this ref? How may I connect it in code?

closed 2017-10-27 by ChiefBureaucraticOfficer