ARMmbed/mbed-os #1025: ST Nucleo 411 PWM pin redirects

BlackstoneEngineering, 2015-04-09:
## Problem
When using the PwmOut API the pin is always redirected to pin D13 (aka pwm2/1). All of the following code snippets will output a pwm wave on D13, all these initializations will compile with no error. 

``` c
// doesnt work
PwmOut out(D0); // Doesnt matter what pin you use, All D# and A# will redirect to D13
int main()
{
    out.period(1.0);
    out = 0.5;
    while(1);
}
```

Oddly though the LED's will map correctly. The following snippet does map to LED1
``` c
PwmOut out(LED1);
int main()
{
    out.period(1.0);
    out = 0.5;
    while(1);
}

```

## Acceptance Criteria
* PWM's map to pin used to initialize object
* board image is fixed to reflect which pins work and which do not.

ohagendorf, 2015-04-09:
I can't reproduce this behaviour. D0 and D1 can't be used as PWM pins because they are SERIAL_TX/RX. To use them the serial port has to be closed.
But e.g. D2 (PA_10) is working well.

Here is my test code:
```
#include "mbed.h"

PwmOut out1(D13);
PwmOut out2(PA_10);
int main()
{
    out1.period(0.01);
    out1 = 0.1;
    out2.period(0.01);
    out2 = 0.2;
    while(1);
}
```
I checked both pins with a scope and got 10ms periods and 1 and 2 ms high pulses.

Which mbed library version did you used? I used the current version sha: 5a81e89

0xc0170, 2015-04-10:
@BlackstoneEngineering Please state the commit you have been using for testing, or at least the mbed library version.

cc @bcostm

BlackstoneEngineering, 2015-04-10:
Using Revision 96 (487b796308b0) of the mbed library, on a St 411 updated to latest firmware. 
According to the pinout diagram on the platform page (http://developer.mbed.org/platforms/ST-Nucleo-F411RE/) pins D0-D15, and A0,A1,A3 can be used as PWM. Ignoring the D0/D1 conflict, I am still observing D2-D15 being redirected to D13. 

Example code can be found here http://developer.mbed.org/users/mbedAustin/code/Scratch/rev/45106b55176e

bcostm, 2015-04-13:
One first thing: if you do this:
<code>
PwmOut out(D6);
PwmOut led(LED1);
</code>

The same timer is used (TIMER2) with channel 3 on D6 and channel 1 on LED1. So the same period will apply on these 2 pins. But this does not explain all the re-directions to D13 observed.

I'll check this also and let you know.

bcostm, 2015-04-13:
I've checked different pwm outputs on D13, D2, D6 and all is ok for me. I don't see any re-direction on D13. I use mbed revision 96.

BlackstoneEngineering, 2015-04-13:
Can you confirm you are using the latest ST Firmware and that you are using the 411 board?

bcostm, 2015-04-15:
Yes I use the NUCLEO_F411RE board but I have the V2.J22.M5 firmware on it. I will try with the latest firmware version and let you know.

bcostm, 2015-04-22:
I have re-checked different PWM with V2.J23.M10 firmware and mbed V97. All is ok on my side. Do you still have the issue ?

bcostm, 2016-05-10:
To be closed

closed 2016-05-10 by 0xc0170