ARMmbed/mbed-os #1029: Tools - Building the mbed library from the copied files

0xc0170, 2015-04-10:
I was told this earlier by @sg- , faced this issue yesterday.

Added one header file to mbed api folder, compiled the mbed library. Later removed the header file from its location, compiled the library with an error. Similar to this - compile a library and inject this to any header file in the build folder for mbed library:

```
#error This header file should be overwritten before compiling
```

And you will see the error present although it was added to the "copied files".

PrzemekWirkus, 2015-04-29:
Can you provide clear steps to reproduce?

BTW, did you consider when we build we are generating .d files with dependency information. Just wild guess, but it may be because of this mechanism ?

sg-, 2017-01-10:
Don't recall more info but we did patch a few bugs with .d logic handling months back. I'll repoen if the problem surfaces again.

closed 2017-01-10 by sg-