ARMmbed/mbed-os #1035: FRDM_K64F - PTA0 vs PTC12 for D8

0xc0170, 2015-04-13:
I reported this a year ago, looks that the newer versions of frdm k64f have it fixed, that D8 is not PTA0 (one of jtag pins), but PTC12. 

In the pinnames, there's D8 as PTC12. Shall we update it, as the PTA0 is unusuable (J11 jumper is shorted by a cut-trace - used by openSDA)

For reference, http://developer.mbed.org/questions/7074/Is-D8-actually-PTC12-and-not-PTA0/

bsilvereagle, 2015-05-18:
Rev D (prior to the fix) is still in pretty wide circulation.

I'm not sure if there is a way to determine the Rev on the fly or not.

bsilvereagle, 2015-06-03:
Just had to bounce around between a Rev D and a Rev E for a project. 

It may be better to have a #define for FRDM_K64F_REVE and FRDM_K64F_REVD for the user to specify which board revision they're using.

0xc0170, 2015-06-03:
Can you send a pull request with your proposal? Thanks

bsilvereagle, 2015-06-07:
Is [the K64F page](https://developer.mbed.org/platforms/FRDM-K64F/) editable to include documentation for this? 

Safe to assume to default to a Rev E, and make Rev <D require the define?

Tinkerer-, 2015-10-07:
Is this still work in progress? I think the picture on the [K64F page] should be updated too. There could be one of those yellow warning boxes below to point people to the difference starting from RevE

0xc0170, 2017-10-12:
I believe this is resolved, as now k64f pinout states PTC12, as it should

closed 2017-10-12 by 0xc0170