ARMmbed/mbed-os #1036: [Export] Multiple definitions error due to libmbed and libc.

EmBlocks, 2015-04-14:
Hi,

Mbed is advertising the syslib 'libc' and also includes the 'libmbed'. The problem is that some libmbed functions are also defined in the libc like 'wait' and 'time' and perhaps more.

I think that the libc shouldn't be on the syslib list, everything builds fine without the libc library.

xiongyihui, 2015-04-16:
@EmBlocks 
I got the same trouble when using EmBlocks.
It seems that it's a problem of the sequence of 'libc' and 'libmbed'. 'libc' should be included after 'libmbed'. 

change `-lc -lmbed` to `-lmbed -lc`

EmBlocks, 2015-04-16:
Do you need that libc or can you also build it without? I have a couple of example projects building fine without lib c but I don't know if this is always the case. 

Do you also have problems with the nano lib option?

xiongyihui, 2015-04-21:
Projects like blinky don't require libc, others may require lib c, so it should be better to keep lib c.

About the nano lib option, do you mean the option `-specs=nano.specs`?  I test `-specs=nano.specs` with Arch Max (STM32F407), it works.

closed 2017-10-27 by ChiefBureaucraticOfficer