ARMmbed/mbed-os #1045: Nordic - Gpio IRQ handler conflict in the mbed lib and nordic-sdk

0xc0170, 2015-04-16:
It looks like there was an update in nordic-sdk, which conflicts with the mbed library for nrf51.

Here are some details:
Ble Thermometer - update all libraries, so BLE_API is 341, mbed lib is 97, nrf lib is 111 (which actually cases this as the gpio app sources were added there). Selected platform - nordic nrf51822 - compiles (haven't checked which gpio irq handler was chosen by linker). Then I remove mbed lib, add mbed-src revs 514 - GPiO IRQ multiple definitions error.

@rgrover

rgrover, 2015-04-22:
we need GPIOTE_IRQHandler to be able to process interrupts arising from GPIO transitions. This handler can't be removed from mbed-src.

GPIOTE is Nordic's meta concept which encompasses GPIO-toogles together with 'tasks and events'. Tasks and events allow applications to capture very short-lived toggles on GPIOs. We don't need this feature.
The solution is to drop the app_gpiote support from nRF51822. mbed-src doesn't need to change.

Updates to the nRF51822 library are currently held up on the next release of the mbed-sdk. As a workaround, it should be safe to drop the files app_gpiote.[ch] from nRF51822.

sg-, 2017-01-10:
@0xc0170 is this still valid?

0xc0170, 2017-01-12:
I havent seen this issue since then, closing

closed 2017-01-12 by 0xc0170