ARMmbed/mbed-os #1046: Ticker class causes SoftDevice Assert and hard fault (nRF51822)

en0b, 2015-04-16:
See details here: https://devzone.nordicsemi.com/question/31378/hard-fault-after-a-while-when-connected/

According to the Nordic support it "should" be okay to disable interrupt for 10-20us and they say if you disable them for more than 200us you are going to have soft device asserts for sure. The Ticker class globally disables interrupts in us_ticker_api.c for about 62us every "tick". In my case this lead to hard faults after staying in a connection for several hours to days.

I think someone should fix this :)

0xc0170, 2015-04-16:
CAn you provide some code snippet which would lead to the hard fault you are experiencing?

cc @rgrover

joseangeljimenez, 2015-04-22:
@en0b Are you using "Replace Nordic's appTimer with a driver for RTC1" https://github.com/mbedmicro/mbed/pull/932?

en0b, 2015-04-23:
@joseangeljimenez no I am not. I was using the normal BLE_API & mbed library releases.

@0xc0170 The code leading to the hard fault is located within the softdevice (ll_lm.s0.c line 1396 with soft device 7.0.0). There is a failing timing assert that is caused by the disabled interrupts in us_ticker_api.c (function us_ticker_insert_event):

```C++
void us_ticker_insert_event(ticker_event_t *obj, timestamp_t timestamp, uint32_t id) { 
60     /* disable interrupts for the duration of the function */ 
61     __disable_irq();  //<<------- Disabled here
62 
 
63     // initialise our data 
64     obj->timestamp = timestamp; 
65     obj->id = id; 
66 
 
67     /* Go through the list until we either reach the end, or find 
68        an element this should come before (which is possibly the 
69        head). */ 
70     ticker_event_t *prev = NULL, *p = head; 
71     while (p != NULL) { 
72         /* check if we come before p */ 
73         if ((int)(timestamp - p->timestamp) < 0) { 
74             break; 
75         } 
76         /* go to the next element */ 
77         prev = p; 
78         p = p->next; 
79     } 
80 
 
81     /* if we're at the end p will be NULL, which is correct */ 
82     obj->next = p; 
83 
 
84     /* if prev is NULL we're at the head */ 
85     if (prev == NULL) { 
86         head = obj; 
87         us_ticker_set_interrupt(timestamp); 
88     } else { 
89         prev->next = obj; 
90     } 
91 
 
92     __enable_irq(); //<<------- Re-Enabled here approx. 62us later
93 } 
```

jrobeson, 2015-10-13:
@joseangeljimenez can you still reproduce this with the lasted mbed releases. this code should be in there by now.

joseangeljimenez, 2015-10-14:
@jrobeson Thanks for the reminder. Unfortunately we have moved on to another project and could not test the latest mbed releases.

seajayshore, 2016-07-12:
This looks the same issue as https://github.com/mbedmicro/mbed/issues/1533

Any progress ever made on solving this?

jrobeson, 2017-07-18:
Is this still an issue these days?

cmonr, 2018-06-04:
Closing this since it's at least 6 months old.

If this is is still an issue, feel free to open a new issue.

deepikabhavnani, 2018-09-11:
> Closing this since it's at least 6 months old.

Closing

closed 2018-09-11 by deepikabhavnani