ARMmbed/mbed-os #1053: [ USBMIDI ] error #321 in USBMIDI.h/MIDImessage.h

K4zuki, 2015-04-20:
I have updated USBDevice to *rev49* on online compiler, I tried to compile but failed by following errors:
* Error: Data member initializer is not allowed in "USBDevice/USBMIDI/MIDIMessage.h", Line: 273, Col: 19
* Error: Data member initializer is not allowed in "USBDevice/USBMIDI/USBMIDI.h", Line: 106, Col: 21
* Error: Data member initializer is not allowed in "USBDevice/USBMIDI/USBMIDI.h", Line: 107, Col: 19

I am sure *rev40* worked ok.
I tested with
* mbed LPC1768
* mbed lib rev97
* USBDevice rev49
* mbed_blinky code
```
#include "mbed.h"

DigitalOut myled(LED1);

int main() {
    while(1) {
        myled = 1;
        wait(0.2);
        myled = 0;
        wait(0.2);
    }
}
```
Could somebody please fix this?

Thanks in advance,
K4zuki

toyowata, 2015-04-20:
May be we need the -u option to kick build.py in build_travis.py.

Fixed here https://github.com/toyowata/mbed/commit/5b74eed7da7af522dd384280bff8d54bc0c9622d and I will make PR once my ongoing PR #1051 will be merged.

0xc0170, 2015-04-20:
Yes, go ahead  with -u.

The PR referenced above was merged, should be fixed.

closed 2015-04-20 by 0xc0170