ARMmbed/mbed-os #1055: AsynchSerial::send should have an overload taking a reference to array

porkybrain, 2015-04-20:
`serial.send(someHandler,"hello world \r\n",15);` seems error prone, relying on programmer discipline to get the length right is not needed.
 
This is also possible `serial.send(someHandler,"hello world \r\n");` and it entails no run time cost. Just add this to http://developer.mbed.org/users/marcuschang/code/AsyncSerial/file/9127f9fae61f/AsyncSerial/AsyncSerial.h

    template<int I>
    inline void send(send_done_t handler, const char (&buffer)[I]){
        send(handler, buffer, I);   
    }

test case:

    #include "mbed.h"
    #include "AsyncSerial/AsyncSerial.h"

    DigitalOut led(LED1);
    AsyncSerial serial(USBTX, USBRX);

    void sendDone()
    {
        led = 1;
    }

    int main()
    {
        const char test[] = "bleep blork\r\n";
        serial.send(sendDone, "hello world\r\n");
        serial.send(sendDone, test);
    
        for(;;)
        {
        }
    }

By the way I got the length wrong in the first line on purpose to prove my point, did you notice? ;)

0xc0170, 2015-04-20:
@marcuschangarm Look at this, might be interested

I am closing this as this is not related to this repository. Please create an issue in the library (if the issue tracker is enabled, if not - contact developer do to so).

closed 2015-04-20 by 0xc0170

porkybrain, 2015-04-20:
I see [here](https://github.com/PrzemekWirkus/mbed/blob/docs/docs/COMMITTERS.md) that I should be testing on multiple compilers with templated code, which compilers are used? What should I be testing on? I have only used GCC on cortex controller so far and I should have no problems with c++03 templates.

porkybrain, 2015-04-20:
Wow, how stupid am I, sorry for bothering you ;)