ARMmbed/mbed-os #1077: Tools - check for installed modules using imp is broken (python 2.7.6/9)

0xc0170, 2015-04-27:
I am not able to run singletest.py as there's an error reported.  The script calles check_required_modules:

```
def check_required_modules(required_modules, verbose=True):
    """ Function checks for Python modules which should be "importable" (installed)
        before test suite can be used.
        @return returns True if all modules are installed already
    """
    import imp
    all_modules_found = True
    not_installed_modules = []
    for module_name in required_modules:
        try:
            imp.find_module(module_name)
        except ImportError as e:
            all_modules_found = False
            not_installed_modules.append(module_name)
            if verbose:
                print "Error: %s"% e
    if verbose:
        if not all_modules_found:
            print "Warning: Mod
```

Is imp is broken? as I have installed both modules. A quick test:

```
λ python
Python 2.7.9 (default, Dec 10 2014, 12:28:03) [MSC v.1500 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import serial
>>> import imp
>>> imp.find_module('serial')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named serial
```

PrzemekWirkus, 2015-04-27:
I'm checking this issue now, on Python 2.7.8 it looks OK (checked on my machine only):
```
C:\Users\przwir01>python
Python 2.7.8 (default, Jun 30 2014, 16:03:49) [MSC v.1500 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import imp
>>> import serial
>>> imp.find_module('serial')
(None, 'c:\\Python27\\lib\\site-packages\\serial', ('', '', 5))
>>> exit()

C:\Users\przwir01>python
Python 2.7.8 (default, Jun 30 2014, 16:03:49) [MSC v.1500 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import serial
>>> import imp
>>> imp.find_module('serial')
(None, 'c:\\Python27\\lib\\site-packages\\serial', ('', '', 5))
>>> exit()

C:\Users\przwir01>
```

PrzemekWirkus, 2015-04-27:
@0xc0170: Can you show me outcome for this below commands in your system?

```
$ python
Python 2.7.9 (default, Dec 10 2014, 12:24:55) [MSC v.1500 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> print sys.path
```
Should be something like this:
```
['', 'c:\\Python27\\lib\\site-packages\\twisted-13.2.0-py2.7-win32.egg', 'c:\\Python27\\lib\\site-packages\\zope.interface-4.0.5-py2.7-win32.egg', 'c:\\Python27\\lib\\site-packages\\pyocd-0.1-py2.7.egg', 'c:\\Python27\\lib\\site-packages\\ws4py-0.3.2-py2.7.egg', 'c:\\Python27\\lib\\site-packages\\web.py-0.37-py2.7.egg', 'c:\\Python27\\lib\\site-packages\\pycurl-7.19.3-py2.7-win32.egg', 'c:\\Python27\\lib\\site-packages\\pyflakes-0.7.3-py2.7.egg', 'c:\\Python27\\lib\\site-packages\\pathlib-1.0.1-py2.7.egg', 'c:\\Python27\\lib\\site-packages\\mbed_host_tests-0.1.5-py2.7.egg', 'c:\\Python27\\lib\\site-packages\\mbed_greentea-0.0.5-py2.7.egg', 'c:\\Python27\\lib\\site-packages\\git_toc-0.0.1-py2.7.egg', 'c:\\Python27\\lib\\site-packages\\mbed_ls-0.1.6-py2.7.egg', 'C:\\Windows\\system32\\python27.zip', 'c:\\Python27\\DLLs', 'c:\\Python27\\lib', 'c:\\Python27\\lib\\plat-win', 'c:\\Python27\\lib\\lib-tk', 'c:\\Python27', 'c:\\Python27\\lib\\site-packages', 'c:\\Python27\\lib\\site-packages\\FontTools', 'c:\\Python27\\lib\\site-packages\\win32', 'c:\\Python27\\lib\\site-packages\\win32\\lib', 'c:\\Python27\\lib\\site-packages\\Pythonwin', 'c:\\python27\\lib\\site-packages', 'c:\\Python27\\lib\\site-packages\\wx-3.0-msw']
```

0xc0170, 2015-04-27:
```
['', 'C:\\Python27\\lib\\site-packages\\markupsafe-0.21-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\pyserial-2.7-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\pywinusb-0.3.3-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\pyyaml-3.11-py2.7-win-amd64.egg', 'C:\\Python27\\lib\\site-packages\\prettytable-0.7.2-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\nose-1.3.4-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\cheetah-2.4.4-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\hgapi-1.7.2-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\pygithub-1.25.2-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\restkit-4.2.2-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\semantic_version-2.3.1-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\markdown-2.5.1-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\socketpool-0.5.3-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\http_parser-0.8.3-py2.7-win-amd64.egg', 'C:\\Python27\\lib\\site-packages\\ntfsutils-0.1.3-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\pathlib-1.0.1-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\intelhex-1.5-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\jsonschema-2.4.0-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\requests-2.5.3-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\setuptools-14.3-py2.7.egg', 'c:\\code\\git_repo\\github\\valinor', 'C:\\Python27\\lib\\site-packages\\yotta-0.2.5-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\mbed_test_wrapper-0.0.2-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\pyjwt-1.0.1-py2.7.egg', 'c:\\code\\git_repo\\github\\project_generator_mbed_examples\\tools\\project_generator', 'C:\\Python27\\lib\\site-packages\\mbed_tools-0.1.14-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\jinja2-2.7.3-py2.7.egg', 'C:\\Python27\\lib\\site-packages\\colorama-0.3.3-py2.7.egg', 'C:\\Windows\\system32\\python27.zip', 'C:\\Python27\\DLLs', 'C:\\Python27\\lib', 'C:\\Python27\\lib\\plat-win', 'C:\\Python27\\lib\\lib-tk', 'C:\\Python27', 'C:\\Python27\\lib\\site-packages']
```

Pithikos, 2015-05-28:
This probably have to do with the fact that `imp` can't find unzipped egg files. So the case might be that `pyserial` is installed but not unzipped for whatever reason. I ran into the same problem some time ago and the solution is in the master branch as seen [here](https://github.com/mbedmicro/mbed/blob/master/workspace_tools/utils.py#L144-170).

So first `imp` is used as before but in the rare case that the module is still an unzipped egg file, `__import__` is used.

sg-, 2017-01-10:
@0xc0170 Is this still a problem? I think we're on 2.7.12 by now

bridadan, 2017-01-18:
I think we're ok to close this.

closed 2017-01-19 by 0xc0170