ARMmbed/mbed-os #1083: Multiple AnalogIn/AnalogOut needs continuous initialization on Nucleo F334R8

bunjes, 2015-04-29:
I want to sample multiple AIn channels and simultaneously output the values to two AOut channels.

The following code works with one ain/aout combination, but if I add the definition of the second AnalogIn/AnalogOut objects, the AOut stops working:

<<code>>
 #include "mbed.h"

int main() {

   // Combination 1
    AnalogOut aout1(PA_5);
    AnalogIn ain1(PC_0);

    // Combination 2
    AnalogOut aout2(PA_6);
    AnalogIn ain2(PC_1);

    while(1) {

        aout1.write_u16(ain1.read_u16());

        aout2.write_u16(ain2.read_u16());

        wait_ms(1); 
    }
}
<</code>>

With both combinations active, I get 0V on PA_5, no matter what is on PC_0, but PC_1 -> PA_6 is still working.

Moving the constructions into the loop partially resolves the problem, but there are artefacts in the output, maybe due to the continuous re-initialization of the DAC/ADC:

<<code>>
 #include "mbed.h"

int main() {

    while(1) {

        // Combination 1
        AnalogOut aout1(PA_5);
        AnalogIn ain1(PC_0);

        // Combination 2
        AnalogOut aout2(PA_6);
        AnalogIn ain2(PC_1);

        aout1.write_u16(ain1.read_u16());

        aout2.write_u16(ain2.read_u16());

        wait_ms(1); 
    }
}
<</code>>

Any help is appreciated.

LMESTM, 2017-06-12:
I'm working on it - a PR should be available this week

LMESTM, 2017-06-16:
@bunjes - See PR #4529 - should solve your issue - feedback welcome

LMESTM, 2017-07-03:
@bunjes  @0xc0170 
#4529 has been merged so issue now solved - please close (and re-open is any problem)

bunjes, 2017-07-03:
Thanks for solving. Two years and two months are quite a time span...
I hope you didn't work on the solution all this time ;-)

LMESTM, 2017-07-03:
@bunjes I'm afraid this issue went through the net for all this time ... we're now cleaning up in a more systematic way to avoid this happens again  ;-)

jeromecoutant, 2017-07-12:
ST_TO_BE_CLOSED

closed 2017-07-12 by 0xc0170