ARMmbed/mbed-os #1096: STM32F4: HSE timeout increased from 0.5s to 5s, needs 10seconds to fallback to HSI

giulianodianda, 2015-05-07:
Hi,
i saw mbed-src have been upgraded to CubeF4 drivers v1.5, which increases HSE timeout from 500 to 5000 ms.
https://github.com/mbedmicro/mbed/blob/master/libraries/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/TARGET_NUCLEO_F401RE/stm32f4xx_hal_conf.h#L104
I don't know if there have been issues with 500ms timeout (afaik issues were about LSE on Nucleo boards with not correct capacitors), anyway now it takes 10 seconds to fallback to HSI (HSEwBypass->HSEwoBypass->HSI).

Sissors, 2015-05-07:
5/10 seconds is indeed way too long for it to be useful. I consider 500ms already very long. With bypass mode there is no reason for anything to startup (unless your oscillator still needs to startup), and it should be nearly instantaniously detected. 

The typical startup time with a crystal is specced at 2ms. Sadly no max is given, but still with that information significantly shorter than 500ms should be no problem, and longer for sure is not required.

0xc0170, 2015-05-11:
Is the header file from the driver package? If yes, this should be asked in their community. I recall @bcostm redirected us to their community for questions related to their drivers.

bcostm, 2015-05-12:
Yes in the Cube package the default timeout is set to 5 sec. I remember that I have changed it manually when I put the Cube package on mbed to 500ms. Maybe it is still too high and we can put a lower value. But recently someone has updated the CubeF4 driver and didn't change this timeout... This is why...

ohagendorf, 2015-05-12:
I'm very sorry about this. I thought that the CubeF4 can be used without any changes except the NULL/HAL_NULL problem. @bcostm, are there more changes you did manually?

We needed the update because we are already using the new STM32F446 mcu which was introduced with CubeF4 driver v1.5. We've created an inofficial NUCLEO F446 (removed the original mcu and resoldered the F446 mcu) and are having a running mbed target. If @bcostm likes to have this target, I can send a pull request with it. Beside this an own board is in development.

bcostm, 2015-05-13:
There is no problem @ohagendorf  :-)

What I did usually when I put a Cube driver on mbed:
- Replace NULL by HAL_NULL. But in the latest Cube driver version (for example CubeF4 v1.5.0) the NULL definition has been removed from the xxx_hal_def.h file. So there will be normally no more an issue (the default NULL equals to 0 is used). For safety this will need to be checked when a new Cube driver is put on mbed.
- Decrease the different timeouts (mainly HSE, LSE).
- Remove the __IO in front of the constant tables (i.e. AHBPrescTable[] in system_xxx.c file. I have sent a request to remove it in the original Cube version.
That's it.

For the NUCLEO_F446RE (I suppose it is this one) it is great that you have already developped it. It is in my todo list but I didn't start yet. And yes, you can publish it on GitHub. I will use it for testing the board as soon as I have it in my hand.

Thanks.

Sissors, 2015-10-31:
Can this be fixed finally? The timeout is ridiculous and it is extremely annoying having to wait 10 seconds after every reset. Not to mention it makes waking up from deepsleep a bit stupid.

star297, 2015-11-01:
It would be better to do away with the auto crystal detect code altogether and use the same format for all the STM platforms as employed for the L476 here. Most if not all other vendor platforms use this manual method. 

#define USE_PLL_HSE_EXTC (0) // Use external clock
#define USE_PLL_HSE_XTAL (1) // Use external xtal
#define USE_PLL_HSI      (0) // Use HSI/MSI internal clock (0=MSI, 1=HSI)
#define DEBUG_MCO        (0) // Output the MCO on PA8 for debugging (0=OFF, 1=SYSCLK, 2=HSE, 3=HSI, 4=MSI)

I have set my L476 to external crystal, this initialises extremely quick and wake up from sleep or deepsleep is fast. The user can also choose the set up if the crystals are fitted depending on their application.

0xc0170, 2015-11-02:
@Sissors what boards are still affected?

Sissors, 2015-11-02:
At least the F4 boards, but I don't know if others are too.

I do like the automatic selection, it is a neat way to solve the problem, however it pretty much makes deepsleep useless (since afterwards it runs that same clock setup code again, instead of remembering what it did on boot), and 10 seconds is just a ridiculous length of time to see if there is an external clock (shouldn't take more than microseconds) and a crystal (which is 2ms typical according to datasheet, but with large spread).

0xc0170, 2015-11-02:
@bcostm @dbestm @adustm Please look at the report above from Erik

dbestm, 2015-11-02:
We think to put 200ms for all products.
OK for you?

Sissors, 2015-11-02:
I think for startup that is fine.

However while it wouldn't be trivial, I think you would still need to reconsider how it behaves after waking from deepsleep. With twice 200ms timeout (once for external oscillator, once for crystal), it takes another 400ms after the wake event happened, before it will be processed (unless all processing happens in the interrupt loop).

dbestm, 2015-11-02:
By default there is no timeout (the external oscillator is check first).
For people who are developing their own code by using sleep mode, it will be easy to modify the clock define to avoid this.

Sissors, 2015-11-02:
In that case my board must have an issue, since it takes 10 seconds to startup.

dbestm, 2015-11-02:
By default SB16 and SB50 (back of the board, where it's written "MCO") should be fit.
Maybe you have a wrong version of the board where it is not done.

star297, 2015-11-02:
SB16 and SB50 are closed if MCO output of ST-LINK MCU is used as input clock.
If X3 (8MHz crystal) is fitted (with associated loading capacitors) SB16 and SB50 must be o/c.

I can only comment from my prototyping point of view. What is needed Ideally is a similar set up to the L476, manual clock set up. However if possible a 'default' set up for auto detect should be added. That way the user can choose and would accommodate for all the variations of the Nucleo that have been produced.   
From experience the L476 set up works very well and is ideal. Please do not change this!

giulianodianda, 2015-11-02:
I can't recall exactly, but i remember some of the reasons why i choosed to not use MCO from STlink:
1- to have more GPIO
2- to be able to use external VCC without having the stlink to pulse the led, which alters ADC reads

Willem23, 2015-11-02:
I think the default auto-setup should be maintained for the average user.
An option to overrule the auto setting is fine, but it should still work with the compiled lib and not force you to use the mbed-src.
It is unfortunate that the first generation of nucleos did not have the MCO bridges installed by default (not sure about the current boards). I have installed the SB16/SB50 bridges on all my boards and that works fine. You also get a higher core clockrate than for the HSI.
The HSI is also notably less accurate than external Xtals which shows in baudrate and other timed signals. You will need board specific corrections to fix that for HSI.

There is an additional issue apart from the Deepsleep mentioned by Erik.
In case you want to use the USB device (or host) features the core clockspeed has to have certain values otherwise you can not get the correct USB clock of 48 MHz. On the F103 for example you can run at 72 MHz core speed with 8MHz external clocks, but you can only get maximum of 48 MHz  core speed using the HSI. It would be nice to include #define settings for USB also.

star297, 2015-11-02:
Having the default auto set up using the MCO STLink MCU 8 MHz clock source is fine if you are only going to use the development board on Mbed official libs and clearly should be employed here.
I think you will still have the timeout issue,  haven't tried, I always use external crystals.
IRC set up is as much use as a chocolate fireguard for anything but the most basic applications.  
Many users will move their code to prototype boards that will not have MCO clock source available. So ideally we need a selection of clock sources and set up's in SRC. Freescale targets have this facilitated and is a good example of what should be available across all platforms to keep a more 'standard' MBED where possible. I'm sure it would be a simple task for the ST guys to do this as they know their libs and they have done this, or similar for the L476.

This is important for low energy applications particularly as ST promotes 'Low Power' here on Mbed.
We need different clock set up's to help facilitate this and to achieve Wim's correct USB clock source.

In the F401 system_stm32f4xx.c we have,

/* Select the clock sources (other than HSI) to start with (0=OFF, 1=ON) */
#define USE_PLL_HSE_EXTC (1) /* Use external clock */
#define USE_PLL_HSE_XTAL (1) /* Use external xtal */

I'm assuming that the default is HSE, so in stm32f4xx_hal_conf.h I have set, 

HSE_STARTUP_TIMEOUT    ((uint32_t) 0) that's zero timeout.
This works providing the HSE crystal is fitted, start up and wake up is then as per spec, around 2mS. Everything works, at least that's what I have found.

0xc0170, 2015-11-03:
> In case you want to use the USB device (or host) features the core clockspeed has to have certain values otherwise you can not get the correct USB clock of 48 MHz. On the F103 for example you can [...]

@Willem23 Please create an issue for this

Willem23, 2015-11-03:
@0xc0170 commented 12 hours ago
@Willem23 Please create an issue for this
Done, see #1408

star297, 2015-12-04:
Guys, could we move this on to finalise.
Can I make a suggestion here, the L476 I believe does have a favourable approach.

https://developer.mbed.org/users/mbed_official/code/mbed-src/file/a11c0372f0ba/targets/cmsis/TARGET_STM/TARGET_STM32L4/TARGET_NUCLEO_L476RG/system_stm32l4xx.c

If you look at the code starting at line 134, this gives a good solution, it makes it simple for the user to select the desired clock source. I have fitted the 8MHz crystal on my board and selecting the 'Use external xtal' does work.

Adding a default for the Nucleo's:
#define USE_PLL_AUTO (1) // Auto detect clock source
Would then cover off the problem with different boards that have been produced and will work 'Out of the Box'.

At line 420 there is a clock set up choice that also indicates how the USB clock will be affected.
Something like this could be the answer.

Moving this set up code into one area or at least one file is also desirable.
A logical format would be:
Define clock source.
Define timeout.
Define clock set up.

0xc0170, 2015-12-08:
cc @dbestm @bcostm

jeromecoutant, 2017-06-01:
Hi
Back to this old issue.
2 things:
- please have a look in #4421
 - as default timeout value in ST official cube is now 100ms (instead of 5s), I will raise a pull request to align our files
Regards,

PR #4422 (2017-06-26): STM32F4 set HSE timeout value to 100ms
## Description
HSE timeout value is set to 100ms
This is aligned with official ST cube since v1.11.0

This fix #1096 

## Status
READY
closed 2017-06-26 by theotherjimmy