ARMmbed/mbed-os #1104: Offline compilation using workspace_tools

yoonghm, 2015-05-13:
1.  Exported files from the online IDE and `workspace_tools\project.py` have very different `Makefile`, specifically `SYS_OBJECTS`, `INCLUDE_PATHS` and `LIBRARY_PATHS` and `CC_FLAGS`. I did not test the exports projects as my local setup is not yet complete. __What is the reason for two different type of Makefile?__ For offline compilation, the user should have the complete mbed file directory such that he/she could get the latest update easily and multiple projects can share the same source code.

2.  I am able to test those _test programs_ using `workspace_tools\make.py`. I noticed that the script is able to detect and compile newly updated files - a feature provided by `make`. However there is no make in my environment. From my initial browsing through the code, it does not use `Makefile`. __How can I use any of the scripts in `workspace\` to compile a non-test programs?__

PrzemekWirkus, 2015-05-28:
You can't.

yoonghm, 2015-05-28:
The reason?

PrzemekWirkus, 2015-05-29:
Just an additional explanation:
You: How can I use any of the scripts in workspace\ to compile a non-test programs?
Me: You can't.

We do not support 'non-test projects' offline compilation with workspace_tools.
Workspace_tools and "python build system' allows you to build:
* mbed SDK library (e.g. to .a file),
* supporting libraries like USBHost/Device / RTOS / CppUTest...
* build tests:  
  * include as dependency mned SDK lib.
  * include as dependencies supporting libs and link with it.
Because:
* We store offline meta-data about test (project) in tests.py & libraries.py.
* If we want to create 'any' project and use offline tools to build it you should probably create extra meta-data for this project similar to entries in tests.py and force

Note: 'offline' we call tests 'projects'. We can export offline test to a Makefile, Eclipse, Keil and other IDEs offline too.

Your apps, components, projects you should create e.g. online using mbed.org. There is logic to put thing together and build you a project you can use.

Note: still nothing stops you from offline building of mbed SDK to lib using build.py script and make it a part of your project's Makefile where you compile and link your app against mbed SDK sources / lib file.

PS: I'm not trying to say building offline non-test projects is a bad idea, we just do not have this thing implemented. Build API's logic in offline workspace_tools is just trimmed to build lib(s) and corresponding tests with it.

bsilvereagle, 2015-06-03:
> we just do not have this thing implemented

What do you mean by that? I've been successfully building a testcase using project.py, deleting any code I don't want & adding any libraries I need, then using those projects successfully for offline compilation of my code.

mfiore02, 2015-06-12:
While it's technically not implemented, all you have to do is add a new "test" entry to workspace_tools/tests.py that points to your project directory.  I've done it dozens of times and it works great.

jrobeson, 2015-10-25:
i wonder if this is worth doing, since yotta can build the software just as easily?

sg-, 2017-01-10:
There have been a lot of improvements to the tools for modified file detections etc. Also mbed CLI is the supported interface to mbed OS tools

closed 2017-01-10 by sg-