ARMmbed/mbed-os #1113: ERROR: run_cmd in build.py

toyowata, 2015-05-17:
I got following error in build.py, even I didn't change my private_settings.py and build toolchain directory.

```
C:\Users\toywat01\Documents\GitHub\mbed>workspace_tools\build.py -m LPC1768 -t ARM -c
Building library CMSIS (LPC1768, ARM)
Copy: core_ca9.h
Copy: core_caFunc.h
Copy: core_caInstr.h
Copy: core_ca_mmu.h
Copy: core_cm0.h
Copy: core_cm0plus.h
Copy: core_cm3.h
Copy: core_cm4.h
Copy: core_cm4_simd.h
Copy: core_cmFunc.h
Copy: core_cmInstr.h
Copy: cmsis.h
Copy: cmsis_nvic.h
Copy: LPC17xx.h
Copy: system_LPC17xx.h
Copy: LPC1768.sct
ERROR: run_cmd: Command 'C:\Keil\ARM\armcc\bin\armcc' can't be found
```

By investigation this issue, I found that it came up by change of utils.py script here.  To workaround of this, comment out the single line below.

https://github.com/mbedmicro/mbed/blob/master/workspace_tools/utils.py#L36

This modification had been done by:
https://github.com/mbedmicro/mbed/commit/ecc0e324409740bfd8bdfda07eb48eb261e6da49
https://github.com/mbedmicro/mbed/commit/62a10cee03515d8f0ece843a889d036fd02b860e

@Pithikos, how did you this this script?

I am using Windows 7 and python 2.7.9.

ohagendorf, 2015-05-17:
Please see PR #1114

0xc0170, 2015-05-18:
I merged 1114, it's working now. I'll close this.

closed 2015-05-18 by 0xc0170