ARMmbed/mbed-os #1123: Problem reading ADC on Nucleo L152RE after deepsleep

hebryn, 2015-05-20:
I have a Nucleo L152RE where I use the library WakeUp to wake the system from deepsleep.

Reading the ADC in the main loop after the call to deepsleep, returns 0.0000. This is not correct, but if I use sleep the value is correct.

Does anyone have a clue how this can be fixed?

Håkon

Sissors, 2015-05-21:
Since mbed forums are bugged (again/still), I'll respond here: The f030 and F401 does not have this issue, so I cannot help you with it further.

However they both seem to take ages to get their clocks running, both from RTC and regular one. Where the F401 manages to take up to a minute between when the interrupt which wakes it up fires, and when the actual main program starts running again.

sg-, 2017-01-10:
@hebryn if this is still a problem please reopen with a test case pointing at where this is broken in the HAL

closed 2017-01-10 by sg-