ARMmbed/mbed-os #1134: New mbed library features implementation

Sissors, 2015-05-23:
This is not a real issue, but a question + a complaint (which you could also call an issue).

To start with: What is mbed HQs plan with the newly introduced APIs? For which targets are you going to implement the async options. What is the time scale for that? What about the other boards? And maybe add on the site an automatically generated table to list which boards support which features.

Then the complaint: Was it ever considered to discuss the new API stuff with the people contributing to mbed? Maybe they have ideas too... I am sure the guys at Silicon Labs had their say, but what about everyone else? Both the hobbyists and developers of other companies? 

Also maybe some documentation would be useful. Random example: https://developer.mbed.org/users/mbed_official/code/mbed-src/file/1abac31e188e/hal/dma_api.h. I am sure I can figure out what the functions are supposed to do, but it is all a bit cryptic now.

Sissors, 2015-05-23:
Looking at the SPI async for example, trying to understand what everything should be for, my conclusion is that it is sadly not suitable at all for SPI LCD displays. Which is really a shame since they are one of the prime candidates for async SPI, since you have alot of data to send there.

closed 2015-05-23 by Sissors

0xc0170, 2015-05-26:
Thanks Erik for sharing and raising this, although you removed your original comments.