ARMmbed/mbed-os #1136: New mbed library features implementation

Sissors, 2015-05-26:
Decided to post this again: This is not a real issue, but a question + a complaint (which you could also call an issue).

To start with: What is mbed HQs plan with the newly introduced APIs? For which targets are you going to implement the async options. What is the time scale for that? What about the other boards? And maybe add on the site an automatically generated table to list which boards support which features.

Then the complaint: Was it ever considered to discuss the new API stuff with the people contributing to mbed? Maybe they have ideas too... I am sure the guys at Silicon Labs had their say, but what about everyone else? Both the hobbyists and developers of other companies? 

Also maybe some documentation would be useful. Random example:  https://developer.mbed.org/users/mbed_official/code/mbed-src/file/1abac31e188e/hal/dma_api.h. I am sure I can figure out what the functions are supposed to do, but it is all a bit cryptic now.

Sissors, 2015-05-26:
For example the SPI async API is in its current form not suitable for SPI TFT displays since it is not possible to make a transaction which sends a fixed value (well unless you only want to send 0xFF).

Personally I also wonder if you really need a seperate RX and TX length argument. The rare case where you want this you can also just split it in two transmissions. The advantage of this is that it is simply easier to understand, and easy to use is (should be) one of the major selling points of mbed, otherwise I simply switch to programming it all directly to the hardware. Also does anyone have a clue what the last parameter is intended for? 

    virtual int transfer(uint32_t *tx_buffer, int tx_length, uint32_t *rx_buffer, int rx_length, const event_callback_t& callback, int event = SPI_EVENT_COMPLETE)  {

Now I am not saying it should be done my way (it should :P), I am saying you should discuss stuff like this with the community. Make a forum topic about it before adding already implementations. That way you also are more likely to get actual implementations. Do you remember the Serial flow control? The current status:
[image]

Another point: I saw the new SPI async API has ability to queue commands. I was really very enthousiastic about this. It is really handy if you can just send all your commands to your LCD and let the library handle it all. Meanwhile you can do other stuff. Then I realised that it was missing one wire to do that: Without chip select functionality built-in that is a bit useless.

anpilog, 2015-05-27:
Hi guys,

My two cents...
We use Serial flow control intensively with no major issues.
Lack of CS support in SPI class is really annoying, however we just built our own SPI class with all required functionality and it works great.

Regards

Sissors, 2015-05-27:
Problem with Flow Control is not that it does not work, it is that it is only implemented for very few boards. The point of mbed is that you have common code for different boards, that is not the case if APIs are simply not implmeneted for most boards.

Currently I also don't have an issue with no CS support included, it is just as easy to do it with DigitalOut. But with async SPI it becomes a completely different story.

anpilog, 2015-05-27:
True 8)

sg-, 2015-05-28:
I know there are lots of 'extended' libraries that bind additional functionality to the core SDK. This isn't the thread for discussing it but think we should think ways to bring this into the SDK or extend the SDK where appropriate.

sg-, 2015-05-28:
and @Sissors would be interested to see an alternative API that works for LCDs. What did you have in mind?

Sissors, 2015-05-28:
I agree those extended libraries could have a look at if they are suitable to be integrated. As random example (because it fits well), a FastPWM has the functionality mbed has, with a better accuracy/wider range, and a smaller code base (not just because I let the mbed lib do the init() part, but also because it does all the common code in a common file, while the mbed code still has alot of copy pasting for PWM).

Regarding alternative API, since I also had some discussions about this with @0xc0170 I made following example: https://developer.mbed.org/users/Sissors/code/spiapi/. Here it would both be capable of sending a buffer, and sending a fixed value, possibly in multiple 8-bit packets if it is set for 8-bit mode. (Templates made it less work, could of course also be functions for 8/16/32-bit).

Now I had some discussions with @0xc0170 and from that it followed that it might be easier to stick to only adding the option to send a single fixed value (what is already does when reading). The remaining issue then is that the KL25 for example only has 8-bit SPI, so for the 16-bit the average TFT uses you cannot send a single fixed value, you need to send two different fixed values. The solution for this could be to simply add 16-bit mode in the SPI driver and implement it by sending two 8-bit packages. This way both for the API and for the user it would appear if it supports 16-bit mode.

And finally I wonder if seperate lengths for TX/RX are required. That is of course not a major issue, but I do think it should be important to keep it clear and easy to use, otherwise you might as well use KSDK functions :P.

sg-, 2017-01-10:
@Sissors regarding original issue, complaint heard. I'll be getting the project part of this repo setup such that there is better visibility and until then, all new api and work can be and is done on feature branches. https://github.com/ARMmbed/mbed-os/branches/all?utf8=%E2%9C%93&query=feature_

Should see a lot of new labels and sorry in advance for all the email spam.

closed 2017-01-10 by sg-