ARMmbed/mbed-os #1152: The latest KDS with K64F export - functional?

0xc0170, 2015-06-03:
This issue is to verify if ksd template is still functional. There has been some reports, some older even ,that KDS export for K64F is not functional.

https://developer.mbed.org/questions/4036/Exporting-to-Kinetis-Design-Studio/?c=18157

We should get the latest version and test K64F, might require update of the template.

bsilvereagle, 2015-07-11:
Using workspace_tools/project.py ( 6cb7294c83 ) I successfully created a kds project and imported it into a fresh install of KDS 3.0.0 for Windows.

0xc0170, 2015-07-13:
@bsilvereagle Thanks for sharing

gjt211, 2015-10-08:
Hi bsilvereagle,
Would you be able to give me a quick rundown on how to get started with mBed for the K64F in KDS?
I have been trying for weeks without great success..
I have installed; KDS 3.0.0, KSDK 1.3.0, and have tried using the template from https://github.com/bundgus/K64F-mbed-KDS-Template
Any advice would be greatly appreciated.

bsilvereagle, 2015-10-09:
@gjt211 I haven't played around in a few months, I'll see if I can get you a write up this weekend. 

To start with, you won't need the KSDK. 

In the meantime, clone this repo, get Python installed, and see if you can get build.py to run.

sg-, 2017-01-10:
KDS should be a supported exporter.

closed 2017-01-10 by sg-