ARMmbed/mbed-os #1177: uvision export doesn't include rtos & rtx source files

mfiore02, 2015-06-10:
The exporter doesn't include the rtos source files when exporting a program using rtos to uvision.  I tested this with the RTOS_1 program and NUCLEO_F411RE.  The export succeeded, but the build in uvision4 failed:

linking...
.\build\RTOS_1.axf: Error: L6218E: Undefined symbol rtos::Thread::wait(unsigned) (referred from main.o).
.\build\RTOS_1.axf: Error: L6218E: Undefined symbol rtos::Thread::Thread(void(*)(const void*), void*, osPriority, unsigned, unsigned char*) (referred from main.o).
Not enough information to list image symbols.
Finished: 1 information, 0 warning and 2 error messages.
".\build\RTOS_1.axf" - 2 Error(s), 2 Warning(s).
Target not created

The rtos and rtx source files were nowhere to be found.

ohagendorf, 2015-06-11:
Did you make a build.py ... -r before?

0xc0170, 2015-06-11:
As @ohagendorf  pointed out above. The script does not build required libraries, you have to do it prior exporting. Does that help ?

mfiore02, 2015-06-11:
Maybe I'm confused.  I thought the exporter (project.py) exported source code for the mbed and other libraries unless the -b option was used.  If the source code is getting exported by default, why would I have to build first?

I've done the following:

python workspace_tools/build.py -m NUCLEO_F411RE -t GCC_ARM -c -j 0 --rtos
............
Build successes:
  * GCC_ARM::NUCLEO_F411RE

python workspace_tools/project.py -m NUCLEO_F411RE -i uvision -c -n RTOS_1
.............
Successful exports:
  * NUCLEO_F411RE::uvision	/home/mfiore/work/mbed_devel/mbed/build/export/RTOS_1_uvision_NUCLEO_F411RE.zip

cp /home/mfiore/work/mbed_devel/mbed/build/export/RTOS_1_uvision_NUCLEO_F411RE.zip .
unzip RTOS_1_uvision_NUCLEO_F411RE.zip
find RTOS_1/ -name "*.c" -o -name "*.cpp"
RTOS_1/env/test_env.cpp
RTOS_1/main.cpp
RTOS_1/mbed/targets/hal/TARGET_STM/TARGET_STM32F4/port_api.c
RTOS_1/mbed/targets/hal/TARGET_STM/TARGET_STM32F4/rtc_api.c
RTOS_1/mbed/targets/hal/TARGET_STM/TARGET_STM32F4/spi_api.c
RTOS_1/mbed/targets/hal/TARGET_STM/TARGET_STM32F4/mbed_overrides.c
RTOS_1/mbed/targets/hal/TARGET_STM/TARGET_STM32F4/analogout_api.c
RTOS_1/mbed/targets/hal/TARGET_STM/TARGET_STM32F4/serial_api.c
RTOS_1/mbed/targets/hal/TARGET_STM/TARGET_STM32F4/pwmout_api.c
RTOS_1/mbed/targets/hal/TARGET_STM/TARGET_STM32F4/gpio_api.c
RTOS_1/mbed/targets/hal/TARGET_STM/TARGET_STM32F4/analogin_api.c
RTOS_1/mbed/targets/hal/TARGET_STM/TARGET_STM32F4/us_ticker.c
RTOS_1/mbed/targets/hal/TARGET_STM/TARGET_STM32F4/i2c_api.c
RTOS_1/mbed/targets/hal/TARGET_STM/TARGET_STM32F4/TARGET_NUCLEO_F411RE/PeripheralPins.c
RTOS_1/mbed/targets/hal/TARGET_STM/TARGET_STM32F4/pinmap.c
RTOS_1/mbed/targets/hal/TARGET_STM/TARGET_STM32F4/sleep.c
RTOS_1/mbed/targets/hal/TARGET_STM/TARGET_STM32F4/gpio_irq_api.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_dma2d.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_i2s.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_i2c_ex.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_gpio.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_iwdg.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_sai_ex.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_rcc_ex.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_eth.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_cec.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_can.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_sd.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_pwr.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_ll_usb.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_rtc.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_i2s_ex.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_rtc_ex.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_dcmi.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_tim_ex.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_adc.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_i2c.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_dma.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_flash_ramfunc.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_hash_ex.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_dma_ex.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_smartcard.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_pcd_ex.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_dcmi_ex.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_ll_fsmc.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_spi.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_spdifrx.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_nor.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_fmpi2c.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_tim.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_ll_sdmmc.c
RTOS_1/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F4/stm32f4xx_hal_dac.c
[... 75 lines cut ...]
RTOS_1/rtos/MemoryPool.h
RTOS_1/rtos/Mail.h
RTOS_1/rtos/Queue.h
RTOS_1/rtos/rtos.h
RTOS_1/rtos/RtosTimer.h
RTOS_1/rtos/TARGET_CORTEX_M
RTOS_1/rtos/TARGET_CORTEX_M/rt_Event.h
RTOS_1/rtos/TARGET_CORTEX_M/rt_List.h
RTOS_1/rtos/TARGET_CORTEX_M/RTX_CM_lib.h
RTOS_1/rtos/TARGET_CORTEX_M/rt_Mailbox.h
RTOS_1/rtos/TARGET_CORTEX_M/rt_Time.h
RTOS_1/rtos/TARGET_CORTEX_M/cmsis_os.h
RTOS_1/rtos/TARGET_CORTEX_M/rt_Robin.h
RTOS_1/rtos/TARGET_CORTEX_M/rt_System.h
RTOS_1/rtos/TARGET_CORTEX_M/rt_MemBox.h
RTOS_1/rtos/TARGET_CORTEX_M/RTX_Conf.h
RTOS_1/rtos/TARGET_CORTEX_M/rt_Semaphore.h
RTOS_1/rtos/TARGET_CORTEX_M/rt_Mutex.h
RTOS_1/rtos/TARGET_CORTEX_M/rt_TypeDef.h
RTOS_1/rtos/TARGET_CORTEX_M/rt_Task.h
RTOS_1/rtos/TARGET_CORTEX_M/rt_HAL_CM.h
RTOS_1/rtos/TARGET_CORTEX_M/os_tcb.h
RTOS_1/rtos/Thread.h

All of the rtos & rtx headers got pulled in, but not the source files...

ohagendorf, 2015-06-11:
The libraries (rtos, dsp, ethernet, ...) are included only in binary form by project.py. Somewhere in the Python code you would find a comment, that there is some work to do to include also libraries in source code. But this part is not yet done.

mfiore02, 2015-06-11:
Ah, I see.  If that's the case, then I should be finding librtos.a or something similar in the export directory, yes?  I can't seem to find anything that looks like a binary version of the rtos library.

0xc0170, 2016-08-25:
@mfiore02 Is this still relevant?

mfiore02, 2016-08-25:
@0xc0170 I will retest against the latest 5.x revision and get back to you.

mfiore02, 2016-08-25:
@0xc0170 Export seemed OK. The build failed miserably, however.

mbed-os-5.1.0-164-gf3d652b

*** Using Compiler 'V5.05 update 2 (build 169)', folder: 'C:\Keil_v5\ARM\ARMCC\Bin'
Rebuild target 'Unnamed_Project'
compiling test_env.cpp...
../../mbed-os/rtos/rtos/Thread.h(61): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/rtos/rtos/Thread.h(86): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/rtos/rtos/Thread.h(112): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/rtos/rtos/Thread.h(137): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/rtos/rtos/RtosTimer.h(47): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(32): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(38): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(51): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(57): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/features/frameworks/greentea-client/source/test_env.cpp(577): warning:  #111-D: statement is unreachable
../../mbed-os/features/frameworks/greentea-client/source/test_env.cpp: 1 warning, 9 errors
compiling unity.c...
compiling mbed-utest-shim.cpp...
../../mbed-os/rtos/rtos/Thread.h(61): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/rtos/rtos/Thread.h(86): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/rtos/rtos/Thread.h(112): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/rtos/rtos/Thread.h(137): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/rtos/rtos/RtosTimer.h(47): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(32): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(38): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(51): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(57): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/features/frameworks/utest/mbed-utest-shim.cpp: 0 warnings, 9 errors
compiling unity_handler.cpp...
../../mbed-os/rtos/rtos/Thread.h(61): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/rtos/rtos/Thread.h(86): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/rtos/rtos/Thread.h(112): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/rtos/rtos/Thread.h(137): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/rtos/rtos/RtosTimer.h(47): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(32): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(38): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(51): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(57): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/features/frameworks/utest/source/unity_handler.cpp: 0 warnings, 9 errors
compiling utest_case.cpp...
../../mbed-os/rtos/rtos/Thread.h(61): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/rtos/rtos/Thread.h(86): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/rtos/rtos/Thread.h(112): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/rtos/rtos/Thread.h(137): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/rtos/rtos/RtosTimer.h(47): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(32): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(38): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(51): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(57): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/features/frameworks/utest/source/utest_case.cpp: 0 warnings, 9 errors
compiling utest_default_handlers.cpp...
../../mbed-os/rtos/rtos/Thread.h(61): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/rtos/rtos/Thread.h(86): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/rtos/rtos/Thread.h(112): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/rtos/rtos/Thread.h(137): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/rtos/rtos/RtosTimer.h(47): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(32): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(38): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(51): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(57): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/features/frameworks/utest/source/utest_default_handlers.cpp: 0 warnings, 9 errors
compiling utest_greentea_handlers.cpp...
[... cut ...]
../../mbed-os/hal/api/FunctionPointer.h(32): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(38): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(51): error:  #1209: invalid argument to attribute "deprecated"
../../mbed-os/hal/api/FunctionPointer.h(57): error:  #1209: invalid argument to attribute "deprecated"
../../src/main.cpp: 0 warnings, 9 errors
".\.build\uvision5\Unnamed_Project.axf" - 184 Error(s), 5 Warning(s).
Target not created.
Build Time Elapsed:  00:04:42

0xc0170, 2016-08-25:
I suspect thats because of old armcc. The newer one  (not certain now the exact version) does compile ok

mfiore02, 2016-08-25:
Ok, I'll close the issue then.

closed 2016-08-25 by mfiore02