ARMmbed/mbed-os #1190: NUCLEO STM32F401 UART 1200 baudrate not works

Asfimer, 2015-06-17:
When any uart baudrate set to 1200 nucleo can not receive any characters/data from sensors etc by any Rx pin. When both baudrate of nucleo and other device set to 9600 everything is ok.  1200 baudrate is critical for SDI-12 based communications.

Willem23, 2015-07-11:
The problem is caused by the fact that 1200 baud can not be generated at the current setting of the clocksource (PCLK2 or APB2) for UART1 and UART6. That clock is running at the SystemCoreClock of 84MHz. A divider is used to reduce that to the serial port samplingrate which is 16x the baudrate (16x1200=19200). You need a divider factor of 4375 (0x1117) and that wont fit in the available 12bits in the divider register BRR. The 1200 baud is just beyond the edge of possible baudrates at 84MHz. There is no problem for 2400 baud. The code will use the truncated value of 0x117 instead and selects a wrong baudrate. This is means the UART still sees data (the Irq fires) but it can not be received correctly and is just garbled..

The easiest solution is to reduce the frequency of the UART clocksource by dividing it by 2 to 42MHz. That is already done for UART2 which connects to the host PC for printf messages.
Code example, should be part of systeminit:
RCC->CFGR &= ~0x0000E000;  // PPRE2, Bit 15-13
RCC->CFGR |=  0x00008000;  // PPRE2: PCLK2 = AHB / 2

See https://developer.mbed.org/questions/54029/USART-Problem-with-NucleoF401RE/

0xc0170, 2015-07-13:
@Willem23 :+1:  for the write-up and help provided.

0xc0170, 2015-07-13:
cc @bcostm

LMESTM, 2017-05-02:
I think that @Willem23 provided the right answer to the question - can this issue be closed ?

LMESTM, 2017-05-23:
@sg- @0xc0170  can you please close the issue ?

closed 2017-05-25 by 0xc0170