ARMmbed/mbed-os #1197: FRDM K64F PwmOut

sarahmarshy, 2015-06-18:
Seems like there is an issue on this board with PwmOut. No output on logic analyzer. Checked that PwmOut IS working for LPC1768.

matthewelse, 2015-06-18:
Is it just one of the output pins of the K64F or all of them?

sarahmarshy, 2015-06-18:
I can try the rest of the pins. Have tried D6 and D7. D7 there is absolutely nothing. D6 with a period of 1s and pulse with of 0.5s looks like below. 
[image]

matthewelse, 2015-06-18:
Have you tried using the default settings (I think with period 0.02s)?

sarahmarshy, 2015-06-18:
Don't see anything with this 
```C++
#include "mbed.h"

PwmOut pwm(D6);

int main()
{
    pwm.period(0.02f);
}
```

matthewelse, 2015-06-18:
Oh ok it's because you need to set a PWM value:

```cpp
int main()
{
    pwm.period(0.02f); // or whatever
    pwm = 0.6; // a float from 0-1, specifying the duty cycle
}

```

sarahmarshy, 2015-06-18:
[image]

Still seems somewhat off for a 20ms period? Should be on for 12ms, right?

matthewelse, 2015-06-18:
oh sorry I just re-read what you said at the start about this working on LPC1768

matthewelse, 2015-06-18:
What if you comment out pwm.period(0.02f);?

sarahmarshy, 2015-06-18:
[image]

Seems inconsistent with pulses.

matthewelse, 2015-06-18:
I would imagine that one of the SDK guys (@0xc0170 or @bogdanm) might be able to help you with this since it's working with LPC1768 but not K64F - sorry I couldn't be more help

sarahmarshy, 2015-06-18:
PEBKAC

matthewelse, 2015-06-18:
aha well done

closed 2015-06-19 by 0xc0170