ARMmbed/mbed-os #1208: feature: allow users to build static libraries (.ar) using online compiler

mfiore02, 2015-06-22:
Some users, myself included, would find it incredibly useful if we could build static libraries using the online compiler.  There is at least one end-certified, production grade device supported on MBED now and it isn't always feasible to release source code to users for security or compliance reasons.  Allowing users to build a static library using the online compiler would quickly and easily solve some of those problems.

0xc0170, 2015-06-30:
Hello, not certain if this is request should go to the mbed.org.

@screamerbg What do you think ?

mfiore02, 2015-07-06:
I think this feature would be useful both offline and online.  See https://github.com/mbedmicro/mbed/pull/1222 for preliminary implementation.

jamesadevine, 2015-07-17:
All,

I have written a small script that does this, it has mercurial integration also:

https://github.com/jamesadevine/mbed-binary-exporter/

sg-, 2016-09-22:
Makefiles should export to reflect the mbed build system. This is an easy way and modify and extend to create such libs. Makefiles are able to be created for iar, arm and gcc

closed 2016-09-22 by sg-