ARMmbed/mbed-os #1217: Change from hard to softfp for FRDM-K64F (Cortex-M4F)

Jags27, 2015-06-30:
Hi,

I have a posted a question in mbed forum about the change of fpu flag for FRDM-K64F from hard to softfp. This processor is having FPU by default, so It would be good to provide the mbed library with the hard enabled instead of soft.

https://developer.mbed.org/questions/54230/Mbed-library-for-K64F-is-built-with-soft/#answer7952

I have been redirected to raise an issue in this forum.

It seems, the flag is changed from hard to soft based on the following request.
 https://github.com/mbedmicro/mbed/pull/224/files 

My request is to change this flag to hard for the better performance.

adamgreen, 2015-07-01:
I will point out that ``-mfloat-abi=softfp`` does use the FPU for single precision floating point operations.  That flag just tells the compiler to pass floating point function parameters in the integer registers instead of the floating point registers.  When it comes time to do something like adding two floating point registers, it will use the FPU to do it.

Jags27, 2015-07-01:
Hi,

I agree with your explanation on the usage of flag. I did some search in the web and found below link.
From the below link, it looks like softfp takes more cycles though as it uses the integer registers.
https://wiki.debian.org/ArmHardFloatPort/VfpComparison

I am actually linking the mbed library (which is by default built with softfp) with CMSIS RTX library (which is built with hardfp) using gcc compiler. During the linking, the compiler throws an error as the libraries are built with different fpu flags.

My request is that it would be good if your provide the two versions of libraries. One is with softfp and other is with hardfp.

Jags27, 2015-07-06:
I am new to github and do not know how to assign this pull request to particular owner.

Can someone assign this issue to dinau who has changed the hardfp tp soft.

0xc0170, 2015-07-06:
I was not aware rtx is built with hardfp command argument.

If you mention a nickname, they should be notified.
cc @dinau

Jags27, 2015-07-06:
Can you respond to my request @dinau

dinau, 2015-07-09:
Hi, 
I'm going to make the pull request to enable hardfp option in a few days.

Jags27, 2015-07-14:
Hi,
Thanks for creating the pull request. How do i get the update on this pull request ? I mean how do i know that library is available with hardfp ?

dinau, 2015-07-14:
Hi Jags27,  
I've yet to make pull request
because I've investigated this issue 
but I've encountered serious problem at this moment.
If you use gcc optimization option "O0", it seems works well.
But all other options e.g. "O2, O3, Os" dose not works.
(The default option is O2)
I don't know why the reason that it dose not work well so far.

The modification is as follows:
workspace_tools/toolchains/gcc.py : 
line 49 :
change from
```
			self.cpu.append("-mfloat-abi=softfp")
```  
to
```  
			self.cpu.append("-mfloat-abi=hard")
```  

Can anynone help to resolve this issue ?
  
dinau

Jags27, 2015-07-20:
Hi,

Can you describe about the error that you are facing @dinau 

Jagadeesh K

dinau, 2015-07-23:
Hi Jags27,  
mbed sdk tools reference to :   
https://developer.mbed.org/handbook/mbed-tools

After above change I mentioned, I executed as follows:  
```python
 python workspace_tools/build.py -m NUCLEO_F411RE -t GCC_ARM -r -c -v  
 python workspace_tools/make.py  -m NUCLEO_F411RE -t GCC_ARM -n RTOS_7 -d . -v
```
You can change the board name 'NUCLEO_F411RE' to 'K64F'.   
I changed tentatively RTOS_7 sample (libraries/tests/rtos/comsis/timer/main.cpp) to 
```c
#include "mbed.h"
#include "cmsis_os.h"

DigitalOut LEDs[4] = {
    DigitalOut(LED1)
};

void blink(void const *n) {
    LEDs[(int)n] = !LEDs[(int)n];
}

osTimerDef(blink_0, blink);

int main(void) {
    osTimerId timer_0 = osTimerCreate(osTimer(blink_0), osTimerPeriodic, (void *)0);

    osTimerStart(timer_0, 200);

    osDelay(osWaitForever);
}
```
You can also change GCC optimization option to any value by modifying  
workspace_tools/toolchains/gcc.py : line 77.  

Resulting:   
<dl>
	<dt>(1) LED blinking OK </dt>
		<dd>-mfloat-abi=<strong>softfp</strong>  </dd>
		<dd>Optimization: -O2  </dd>
	<dt>(2) LED blinking OK </dt>
		<dd>-mfloat-abi=<strong>hard</strong></dd>
		<dd>Optimization: -O0</dd>
	<dt>(3)LED blinking NG</dt>
		<dd>-mfloat-abi=<strong>hard</strong></dd>
		<dd>Optimiztion: -O1,-O2,-O3,-Os</dd>
</dl>
	

dinau

theotherjimmy, 2017-04-25:
Is this still a problem? RTX is included in mbed OS 5 releases now. @Jags27

theotherjimmy, 2017-05-16:
No response. I'm closing. Please re-open if this is still a problem.

closed 2017-05-16 by theotherjimmy

adamgreen, 2017-05-16:
I just thought I would add some recent notes from when I recently tried switching this ABI flag.

I have recently been able to successfully build all mbed-os code for the K64F using -mfloat-abi=hard. I had temporarily switched to building all Cortex-M4F targets with that flag but that wasn't feasible since there are some targets like MAX32620HSP which link in BPL licensed code that have been compiled with -mfloat-abi=softfp. This leads to link failures because of mismatched ABIs between modules. I just switched back to using softfp for Cortex-M4F parts.