ARMmbed/mbed-os #1223: PR #1139 seems to break singletest automated test if mbed_lstools module is not installed

toyowata, 2015-07-07:
```
>python workspace_tools\singletest.py -c copy -i ..\spec.json -M ..\muts_default.json
Traceback (most recent call last):
  File "workspace_tools\singletest.py", line 66, in <module>
    from workspace_tools.test_api import SingleTestRunner
  File "C:\Users\toywat01\Documents\GitHub\mbed\workspace_tools\test_api.py", line 56, in <module>
    from workspace_tools.compliance.ioper_runner import get_available_oper_test_scopes
  File "C:\Users\toywat01\Documents\GitHub\mbed\workspace_tools\compliance\ioper_runner.py", line 23, in <module>
    import mbed_lstools
ImportError: No module named mbed_lstools
```

0xc0170, 2015-07-07:
cc @PrzemekWirkus

jsunderland, 2015-07-13:
I just did a pull and now I am getting the same issue. Is there a work around for this? I am still new to mbed and I am not sure how I can install the mbed_lstools module.

toyowata, 2015-07-23:
Fixed by https://github.com/mbedmicro/mbed/pull/1259

closed 2015-07-23 by toyowata