ARMmbed/mbed-os #1232: Where is io_cwb defined?

goldscott, 2015-07-10:
In mbed/libraries/USBDevice/USBDevice/TARGET_RENESAS/TARGET_RZ_A1H/usb0/src/userdef/ there are four files that declare function io_cwb as "extern" but nowhere is this function actually defined.

toyowata, 2015-07-10:
The io_cwb() function is called when CACHE_WRITEBACK is defined (and actually it isn't defined).

https://github.com/mbedmicro/mbed/blob/master/libraries/USBDevice/USBDevice/TARGET_RENESAS/TARGET_RZ_A1H/usb0/src/userdef/usb0_function_userdef.c#L57

Are you trying to enable cache writeback?

cc @masaohamanaka

goldscott, 2015-07-10:
Yes, CACHE_WRITEBACK isn't defined, but what if a user wanted to enable
cache writeback? The io_cwb function isn't defined so this cache mode is
currently unusable. I'm not necessarily trying to enable cache writeback,
I'm just notifying the community of issues.

-Scott

TomoYamanaka, 2015-07-27:
Hello,

Thanks for your comment.
We have a plan to modify this problem.
For development status, please refer to the following URL.

https://developer.mbed.org/teams/Renesas/wiki/Known-issues-of-GR-PEACH
No.16 of Section "issues"

0xc0170, 2015-07-27:
@TomoYamanaka thanks for sharing that list, good job !

TomoYamanaka, 2017-02-08:
Sorry for this late reply.

mbed official does not support DMA, so it does not support in USB Function driver for RZ/A1H.
However, if you excute `Cache Invalid` or `Cache clean`, you can refer to the following processing.

https://developer.mbed.org/teams/Renesas/code/GR-PEACH_mbed-os-client-ZXingSample/file/5d3134f9efae/zxing_main.cpp
Line:107 - 128
```
/****** cache control ******/
static void dcache_clean(void * p_buf, uint32_t size) {
    uint32_t start_addr = (uint32_t)p_buf & 0xFFFFFFE0;
    uint32_t end_addr   = (uint32_t)p_buf + size;
    uint32_t addr;

    /* Data cache clean */
    for (addr = start_addr; addr < end_addr; addr += 0x20) {
        __v7_clean_dcache_mva((void *)addr);
    }
}

static void dcache_invalid(void * p_buf, uint32_t size){
    uint32_t start_addr = (uint32_t)p_buf & 0xFFFFFFE0;
    uint32_t end_addr   = (uint32_t)p_buf + size;
    uint32_t addr;

    /* Data cache invalid */
    for (addr = start_addr; addr < end_addr; addr += 0x20) {
        __v7_inv_dcache_mva((void *)addr);
    }
}
```

toyowata, 2017-03-21:
@goldscott

closed 2017-10-27 by ChiefBureaucraticOfficer