ARMmbed/mbed-os #1255: [LPC15xx] SPI::aquire can generate extra clock edges

devanlai, 2015-07-21:
The `SPI::write` method calls `aquire` to ensure that its frequency/mode settings are the current settings in use. However, on the LPC1549, changing the frequency/mode generates a couple of spurious clock edges. If hardware slave select is not used, this can cause the first SPI write that resets the frequency to generate extra clock edges while the slave select is asserted, which can confuse the slave device.

For example, consider the following code run in a loop

    cs = 1;
    spi.write(0x55);
    cs = 0;
    other_cs = 0;
    other_spi.write(0x55);
    other_cs = 1;

This results in the following logic analyzer trace:
[image]

The extra clock edges causes the analyzer to misread 0x55 as 0xD5 because the first two bits are read  while the SPI peripheral is being reconfigured. The actual 8 bits of data are shifted over by two.

In this case, the two SPI objects are using the same peripheral, but the logic in aquire doesn't care if the two peripherals are the same, so this problem could just as easily occur if the two objects used different peripherals.

anpilog, 2015-07-22:
Hi,

Yeah... I faced same problem...
I think proper way to solve it is make aquire() method public.

I usually use two workarounds:
1) When I need to use SPI in the driver code then I inherit driver class from SPI and call aquire() method before push CS down.
2) For main.cpp I have speccial wrapper class that take case about CS and proper aquire() call.

Regards

Sissors, 2015-07-22:
I don't think that that is a proper solution, then depending on how you implement it exactly, every single library in existence needs to be modified to still work, or to work with the LPC1549.

It is a problem which really shouldn't happen in the first place, but okay it happens. Without having the target myself: Possibly it can be solved by switching to DigitalOut (with state which is equal to clock idle state), changing the clock settings, and switching back?

anpilog, 2015-07-23:
@Sissors NRF51 SPI driver has same issue as devanlai described about LPC1549.
STM32 is a bit better but the problem is there anyway. Because aquire() method and especially mode changing must happen when CS is inactive.

Example:
```cpp
#include "mbed.h"

SPI spi1(PB_5, PB_4, PB_3);
SPI spi2(PB_5, PB_4, PB_3);

DigitalOut cs(B5);

void setup()
{
    spi1.frequency(1000);
    spi1.format(8, 0);

    spi2.frequency(1000);
    spi2.format(8, 2);
}

void loop()
{

    cs = 0;
    spi1.frequency(1000);
    spi1.write(0xAA);
    cs = 1;

    cs = 0;
    spi2.frequency(1000);
    spi2.write(0x55);
    cs = 1;
    wait_ms(100);
}
```

This is how it looks on different CPU:
NRF51
[image]

STM32F103:
[image]

STM32F405
[image]

Sissors, 2015-07-23:
Only thing I see for the STM32 is that it droops a bit? It goes maybe into a high-impedance state for short time? For sure it doesn't affect digital there. NRF51 clearly shows indeed the same issue. Still I don't think the issue is forcing users to acquire all the time themselves. Disconnecting the SPI peripheral from the pin when it is being problematic seems to me to be nicer.

anpilog, 2015-07-23:
Well... it's not so good...

Most of SPI slaves could work with STM32 semi-correct spi signals but some would not.
For example serial flash usually ignore such signals due to wrong clock level.
It's also going to be a problem for SPI slaves with autodetecting of SPI mode.

Here is correct signal sequence for SPI mode 0 and 2.

SPI MODE0
[image]

SPI MODE2
[image]

Sissors, 2015-07-23:
Ah I didn't check the waveforms properly.

That said I still don't see a reason to force users to call acquire all the time just because a few targets have some weird hardware, and some others have bad software. Masking the hardware problems (for example by switching it to GPIO when changing frequency), and fixing software issues seems more logical to me. 

Because I assume the reason for the issues which an STM one can have is their code: https://developer.mbed.org/users/mbed_official/code/mbed-src/file/5afbce4ea2da/targets/hal/TARGET_STM/TARGET_STM32F4/spi_api.c, it follows the same coding style as more and more targets seem to be using sadly, the: "Have you try turning it off and on again" style. Where they completely disable the peripheral and reinitialize it when they want to switch mode/frequency (also popular with PWM).

In other words: Users shouldn't be forced to manually call that function because it is implemented poorly: The implementation needs to be fixed.

anpilog, 2015-07-23:
Well.. I don't try to convince you that user MUST trigger aquire() each time by him self 8)

I suspect that case with wrong clock level after CS is valid for any mbed platform because it's logical issue. It has nothing with STM32 or Nordic HAL implementation.

anpilog, 2015-07-23:
I haven't check popular libraries from mbed community but I guess people call format() or frequency() before trigger CS.
This is probably easiest way to workaround this issue.

Sissors, 2015-07-23:
They do, it makes little sense not to do it before triggering CS. However if I understand this correctly the issue is that format and frequency are called by acquire, which is not done before setting CS. Switching the frequency should not cause spurious clock edges. Switching between modes where the clock idle value is different will of course generate a clock edge, that can't be prevented, and for that you could make acquire public. 

However in my opinion any change either in frequency or in phase (so not clock idle level), should not be visible on the output. Otherwise you either got a problem in hardware (which you can cirumvent in software), or you got the problem in software in the first place.

anpilog, 2015-07-24:
True.

I see it as non technical issue but logical.
This issue appears only in applications where SPI slaves work in different modes.
Other cases, for example LPC1549 is just bug in spi_api.c or SPI.cpp.

devanlai, 2015-08-04:
LPC1549-specific
------------------------
After some experimentation, it looks like when SPI is disabled on the LPC1549, it disregards the `CPOL` bit and actively drives `SCLK` low. As @Sissors suggested, it is possible to blank this by temporarily rerouting `SCLK` to a GPIO that just drives the idle clock state as long as SPI is disabled. I tried it out and it works, but it feels like it might be overreaching a bit, having SPI dynamically configure a GPIO temporarily.

The alternative of simply leaving SPI on while reconfiguring it seems better to me. I didn't see any misbehavior while modifying `CPHA` or the frequency. Changing `CPOL` generates an edge when the idle state is redefined, but as previously mentioned, that's unpreventable.

The manual does mention that the `CPHA`/`CPOL` should not be modified unless SPI is idle. I can only see this being a case in slave mode (which would be weird) or when using the new asynchronous API. I'm not sure what should be done if someone tries to use the synchronous API to reconfigure SPI (or read from / write to) while the asynchronous API is running. The only logical behavior I can see would be to wait for all pending transactions to complete, but right now it doesn't look like the synchronous API has any awareness of the asynchronous API for any operations at all.

The other case where the manual recommends doing a full disable/enable is when switching between the master/slave roles. I haven't tried this, but assuming the manual is correct and `SCLK` is only driven if the master role is selected, I don't think we would see any spurious edges when disabling and reenabling it to change roles.

SPI API in general
-------------------------
At the API level, I agree that it would be useful to have make `aquire()` (or its correctly spelled cousin `acquire()`) public.

The way that I wanted to use multiple SPI objects was to have one SPI object per slave device. Each SPI object knows the appropriate frequency/mode to use and automatically reconfigures the port to use it when the object is used, thus avoiding the need to carefully call `format()` and `frequency()` on a single shared SPI object.

The only drawback is that with software slave select, there's no nice way to control when the port is reconfigured; the first SPI operation will have extra delay before the actual transfer because the port needs to be configured. I could call `format()` or `frequency()` to force a call to `aquire()`, but it's a bit silly that I have to repeat the frequency I want and force the peripheral to be reconfigured when it might not even need reconfiguration. Allowing a direct call to `aquire()` would nicely solve this, especially if `aquire()` were to use the new `spi_get_module()` HAL function to avoid unnecessary reconfigurations when different SPI ports are involved.

closed 2017-10-27 by ChiefBureaucraticOfficer